TauLabs/TauLabs #1772: Proposal: remove support for CC (not CC3D).

mlyle, 2015-08-05:
There are very few of the boards out there and probably none are flying.  Supporting CC uses a few bytes of CC3D memory and adds complexity in several places.

@solidgoldbomb @CheBuzz @guilhermito @elafargue @peabody124 any objections?

DTFUHF, 2015-08-05:
This will probably ease confusion too among new TL adopters.

Is CC original even open source HW?

mlyle, 2015-08-05:
It appears to have documents in the tree.

tracernz, 2015-08-05:
This would make it easier to clean up the F1 ADC stuff a bit too.

mlyle, 2015-08-06:
Also on the list: remove support for I2C PCF8591 ADC (only supported on CC flexiport, right now, and doesn't build by default).

peabody124, 2015-08-06:
I don't love the idea if it is just to remove a few switch statements. I believe the big memory hit is the ADC FIFO which is dynamically allocated on CC (meaning it probably doesn't support autotune).

That being said, supporting pretty old hardware with unavailable sensors shouldn't block serious enhancements.

mlyle, 2015-08-06:
So, basically, there's three reasons.  First, the way the sensors is bifurcated makes it a little more complicated to handle non-acro Naze variants in the future.  Second, tracernz is going to do some F1 ADC cleanup for battery monitoring on OpenNaze that it'd be nice to not account for primary flight sensors being on that on CC.

But the really big one is to keep the size of the test matrix sane.  If no one flies it, either it ends up untested or developers spend effort keeping something working that no one uses.  This is different from CC3D where there's a vibrant community around these targets and a large number of initial units.  The number of CC units being flown out there is possibly 0 and is certainly small and rapidly decaying.

SystemsGuy, 2015-08-08:
It does beg the question - is anyone still flying a CC?  I've had mine in a box for at least two years...

guilhermito, 2015-08-08:
Yes I'm still flying it, so I would prefer to continue to support it as a target

mlyle, 2015-08-08:
OK, that's good enough for me-- at least one flying out there.  I'll try again in a couple releases ;)

closed 2015-08-08 by mlyle