TauLabs/TauLabs #1774: RFC: Failsafe++

mlyle, 2015-08-07:
So I know failsafe has been a very controversial topic.

I think the long term solution has to do with improvements to alarms infrastructure and convergence checks where we can figure out if something we're trying to do autonomously is "working" and if it is possibly not to fail safe.

I'd like to suggest an interim step on the way, where we have an intermediate failsafe mode that is specified by the user.  It will have a configurable duration of at most 3000ms (3 seconds), and can be selected between altitude hold, altitude hold max descent, or levelling with a throttle setting not to exceed 30%.  Infrastructure will be added for calling a mode-specific convergence check that will help us later gain confidence to add RTH and PH to the list.

This should have hysteresis, so that the timer doesn't get reset until a long time-- say 5 seconds, without a failsafe to prevent individual frames of control keeping us in the mode indefinitely.

Example-- T=10s we lose control for 1.5 seconds and enter levelling mode at 30% throttle (a speedy descent on most airframes).  T=11.5s we regain control.  T=12.0s we lose control again and re-enter levelling.  T=13.5s the real full failsafe of motor stop occurs.

Thought needs to be given on how this interacts with receiver failsafes/timeouts, PPM timeouts, and other mechanisms.  For instance, on my OpenLRS rig I have effectively done this at the receiver level-- I generate predetermined outputs for a second for levelling and low power before stopping PPM.

mlyle, 2015-08-07:
For clarity-- allow user to set mode, throttle setting (NTE 30%), and duration (NTE 3000ms).  Default off.  Only expose in UAVOs.

fujin, 2015-08-08:
Not sure what I have to offer here apart from this: when I configure my radios with zero-pulses, I expect the machine to fall out of the sky until the radio is re-attached, complete "stop the world". It has by far been the safest for quads, for me. I experimented early on with leveling / low throttle stuff but mostly found it drifting into stuff with the motors running, compared to a full stop, which while the impact force is increased by the height, no additional damage from motors running during failsafe-level-landing occurs. At most you get a can-opener and cameras being thrown away.

mlyle, 2015-08-08:
So the intention here two things.

1. Lay the groundwork for RTH-failsafe eventually.
2. Deal with radio blips better.

I think, given the choice between (say) 0.75 seconds of radio dropout results in everything stopping, and 0.5 seconds of radio dropout results in 1.5 seconds of levelling at low power ... the latter might be safer and provides you with a chance to start getting frames through and save it.

peabody124, 2015-08-12:
See also #1410 for very similar discussion. I totally agree with starting to create this infrastructure. I still think it strongly depends on https://github.com/TauLabs/TauLabs/issues/1220 to be safe. I think the flight status module that @BrainFPV created in https://github.com/TauLabs/TauLabs/issues/1635 could be quite relevant for that