TauLabs/TauLabs #1776: Pirouette Compensation

fujin, 2015-08-08:
Per the old vbar piro comp of yore, we might consider rotating the integrals for pitch and yaw such that during pirouette, no extra rolling action is added. We should also make sure to not do this in leveling/horizon due to potentially adverse side effects.

https://youtu.be/gslSFUj6cnk
https://youtu.be/NDzKVFRrf7Q

OP references: https://github.com/openpilot/OpenPilot/blob/29b9265531a808c8e337c4c9ff0091888bbd89da/flight/modules/Stabilization/innerloop.c#L325-L334
https://github.com/openpilot/OpenPilot/blob/29b9265531a808c8e337c4c9ff0091888bbd89da/shared/uavobjectdefinition/stabilizationbank.xml#L21

Implementation looks rather simple. Thoughts? :heart_eyes_cat:

fujin, 2015-08-08:
:hear_no_evil:

mlyle, 2015-08-31:
Do we still think this has any value?

mlyle, 2015-09-10:
Inactive.  If shown to do something useful could be considered.

closed 2015-09-10 by mlyle