TauLabs/TauLabs #1778: Rate Gyro Suppression

fujin, 2015-08-08:
Leading on some of the code I sketched out in https://github.com/TauLabs/TauLabs/issues/1776 it should be possible to implement something akin to `Acro+`, that is Rate with Gyro Suppression/factor. "Insanity Factor".

fujin, 2015-08-08:
acro insanity factor @ https://github.com/openpilot/OpenPilot/blob/29b9265531a808c8e337c4c9ff0091888bbd89da/flight/modules/Stabilization/innerloop.c#L290

fujin, 2015-08-08:
I'm fairly certain acro+ is just rate+gyro suppress, but would welcome enlightenment.

Trex4Git, 2015-08-08:
That was also my understanding when I had a look at the source code.
I had written in the acro thread on forums also a short impression of Acro+ http://forum.taulabs.org/viewtopic.php?f=12&t=704#p6000, as  I think it may work better then MWRate or VirtualFlybar, as the basic tuning is the same as normal rate mode, and autotune should work.

But as there was no feedback if its worth to implement this, i had no further look until now. I seemed to me that high gyro rates + expo was fine for the acro pilots .

fujin, 2015-08-08:
@Trex4Git Thanks for confirming. You have my feedback now: it is worth it to implement this, and if you see my previous works, it should be relatively simple to implement inside of stabilization_rate.

Trex4Git, 2015-08-08:
;) had no look at your previous works yet, but are they directly related to Acro+ functionality?
I mad no deep analysis of the OP code, but my first impression was, that it should be just a few new lines of code.

Most work might be implementing a new flight mode, or do you think that this should be integrated into normal rate mode?

mlyle, 2015-08-08:
So I really believe that acro+ and mwrate are equivalent.  And it's close to virtualbar.  I don't really like the idea of getting a third way to do the same thing.

Trex4Git, 2015-08-08:
I think virtualbar and mwrate are more similar to each other, then to acro+.
The first 2 try to control to keep the angle constant, and only the stick input with gyro suppression makes it rotating.
Acro+ is for lower stick inputs like normal rate mode, so already rotating with control, and adding than gyro suppression and mixing pure stick input to the output.
So for higher stick input all 3 should be similar, but for lower stick input acro+ might has a different behavior.

But i am still not sure if the difference to normal rate + expo or mwrate/virtualbar is so big, and when the core developer don't like it i am not very motivated to implement it.

mlyle, 2015-08-08:
> The first 2 try to control to keep the angle constant, and only the stick input with gyro suppression makes it rotating.

This is not true.  There is no attitude loop, and the rate loop takes the stick input command.
~~~
  float *raw_input = &stabDesired.Roll;
...
  actuatorDesiredAxis[i] = pid_apply_setpoint(&mw_pid,  raw_input[i] / cfgP8,  gyro_filtered[i], dT);
~~~

Admittedly it is an odd control structure, but the setpoint moves based on stick command.

Trex4Git, 2015-08-09:
Sorry, i don't remembered correctly.

It was virtualbar where the setpoint was 0. In mwrate only the P coefficient is set to zero before the pid function is called, and than there is later a little bit strange p-term added.

mlyle, 2015-08-21:
For consideration, tagging w/ endo release.

mlyle, 2015-08-29:
@peabody124 merged this, so closed.

closed 2015-08-29 by mlyle