TauLabs/TauLabs #1779: USBComBridge: Sending UART->VCP Results in Reboot

tracernz, 2015-08-09:
I am testing on Quanton with the 201507 release with default settings except for:
UART1: ComBridge
USB_VCPPort: ComBridge
Com Bridge module enabled

Sending characters from the VCP port to the UART works fine. Sending a single character from the UART to the VCP port results in the character transmitting successfully and then the flight controller rebooting.

mlyle, 2015-08-09:
So I suspect this could be my regression, so self-assigning.

tracernz, 2015-08-15:
I have narrowed this down to the switch to ChibiOS in eab08607eb476053003c04009faafd90523bd136.

tracernz, 2015-08-15:
I'm quite convinced it's a stack size issue, here are stack free measurements:
749601a43bba46b67fb1ce5afb2d3bf08bab1478 (before chibi), after some data sent, Doesn't crash
176 com2usb free
180 usb2com free
384 max STACK_SIZE_BYTES

eab08607eb476053003c04009faafd90523bd136 (switch to chibi), no data on uart, CRASHES with data
12
16
384

eab08607eb476053003c04009faafd90523bd136 with stack increased to 600, some data sent, Doesn't crash
164
200
600

39fff89856264fc9acf4d0967b69d4ddf25d5979 (current next) no data on uart
44
44
384

You can see the stack usage increased a lot with Chibi..

mlyle, 2015-08-15:
On Sparky2, current next, ComUsbBridge does nothing.  Doesn't reset, doesn't echo chars through a loopback cable.  Tested also with increased stack sizes to no avail.

mlyle, 2015-08-15:
ComUsbBridge tasks both show up as running in task monitor.

mlyle, 2015-08-15:
It's getting to be oscilloscope time.  Confirmed I can receive and transmit on VCP just fine.  So something's wrong with the actual USART communication/loopback.

tracernz, 2015-08-16:
Okay, I have tested current next on Quanton with the ComBridge stack sized bumped up to 600 bytes and the rx->tx lines shorted so the UART just echoes. In the second set of numbers I sent README.MD from the root dir to see how much stack is used when we actually send and receive some data.
```
w/ 600 byte stack, remaining:
before any data sent:
com2usb: 172
usb2com: 260
after sending ./README.MD
com2usb: 156
usb2com: 212
```
So the maximum stack usage seen is 444 bytes. That is quite a bit more than the 384 bytes allocated at the moment. I suggest we bump this to at least 500 bytes, thoughts?

I suspect the Sparky2 not working is a different issue as the module is definitely functioning on the Quanton once the stack is bumped up.

mlyle, 2015-08-16:
I have 480 bytes on the HAL branch now.

mlyle, 2015-08-16:
Sparky2 was not working because CAN always was the combridge target because of some CAN debug code left in.

mlyle, 2015-08-18:
Fixed by the HAL branch #1759 for Sparky2 (CAN override) and other targets (combridge stack size.

closed 2015-08-18 by mlyle