TauLabs/TauLabs #1783: Delete vbar

fujin, 2015-08-09:
Per #1782, consider nuking vbar

fujin, 2015-08-09:
Is vbar still used in old 3d helos? is taulabs used for these anywhere at all?

mlyle, 2015-08-14:
@peabody124 what do you think?  Does it accomplish something for helicopters that is not best done some other way?

tracernz, 2015-08-14:
Virtual fly-bar isn't for old 3d helis, they have an actual physical flybar. This is supposed to make it "feel" more like a flybar heli rather than the more robotic action of flybarless helis, by damping the controller action. I wonder if anyone is actually flying a CCPM heli with TL (or any multicopter controller for that matter) as they are so far behind heli-specific gear in terms of flight performance?

peabody124, 2015-08-14:
If we want to have any pretense of helicopter support we should keep this flight mode. The pirouette compensation in particular is an important part of it. The strange types of limits it places on the virtual swash plate is also good. These features are not present in other modes.

e.g. https://vimeo.com/37255374

These were things I developed by looking at how the align system physically responded and also the phubar code.

mlyle, 2015-08-14:
> If we want to have any pretense of helicopter support we should keep this flight mode. 

Helicopter support or CCPM support?

I also wonder if the pirouette compensation might be better handled by an input transform.

I am nervous about flight modes rotting that no one flies-- they could be badly regressed and we'd never know.

peabody124, 2015-08-15:
It can easily fly on a multi rotor or plane too. I'll test it tonight or tomorrow and make sure it is still good.

mlyle, 2015-09-10:
And indeed there's increasing heli usage that justifies keepin' this around.

closed 2015-09-10 by mlyle