TauLabs/TauLabs #1784: Inner loop critical priority / outer loop modules

fujin, 2015-08-09:
Split loop into inner/outer loop, implement inner/outer loop blending, use common functionality for Horizon and Acro+. If possible try to do all time critical functionality in inner loop and see about synchronizing the inner loop task to Gyro IRQ.

Trex4Git, 2015-08-10:
I think your request is based on the OP inner/outer loop refactor.

OP/LP has not only the task managed from RTOS, but for each task priority there is a additional callback functionality which has highest priority.
This seems also to be used for the inner/outerloop there. I don't know how much advantage this has compared to only tasks priorities from the RTOS (may be an advantage when we don't have enough different task priorities). 

With the the involved task and the data processing the Stabilization task is already synced to the Gyro IRQ, I think. 
But as there might be some priority problems it is worth to do further investigations if there is really to much jitter, and if it can be corrected with the functionality we already have or something new is needed.
And when we really don't wont to run the "outer loop" stuff every time some more time critical "inner loop" stuff is run, a simple clock divider in software would not be a much work.

And I am also not sure if splitting it into 2 tasks/callbacks has a big advantage. So the common functionality between Horizon and Acro+ is just the Rate stuff, and that are just a few lines of code. And you would also have some overhead when splitting it.
But maybe refactoring stabilization.c in a way that not all functionality is directly in the switch/case construct, but we put some common used functionality in sub functions that can be re used, makes sense.

And keeping everything together without splitting seems to me at the moment more flexible, than a split.

On IRC I have some nice ideas for possible enhancements in future, like when there is feedback from ESC with torque information, and a additional control in the rate loop.
Than the fixed construction with inner- and outerloop would not fit anymore. But when keeping them together refactoring might be much easier.

mlyle, 2015-09-10:
I don't think this is likely to be a win.  While we need overall study of performance and task priorities, the current sensors/attitude/stabilization/actuators factoring seems to be sane.  We'd certainly look at other factorings in a PR if they were shown to be more performant or cleaner in some meaningful way.

closed 2015-09-10 by mlyle