TauLabs/TauLabs #1785: expo5 for Rate

fujin, 2015-08-10:
Following on from a conversation on IRC, selectable expo (pow3 vs. pow5) or switch to pow5 for rate expo.

fujin, 2015-08-10:
> <icee> so basically if you're not using an odd power you need to compare if                                                                                                                                                                    
       it's greater or less than 0

fujin, 2015-08-10:
> icee: for odd powers it's .8 x + .2 x^3, say.  for even powers it's .8 x + .2x^2 if x is more than 0,    .8 x - .2 x^2 if x is less than 0

fujin, 2015-08-10:
> .8 x - .2 (-x)^m for x <0

fujin, 2015-08-10:
https://www.desmos.com/calculator/jai5zbn6lt

mlyle, 2015-08-11:
If we change it so that the control exponent m is configurable along with the expo fraction a, the GCS can be adapted to take 3 parameters:

* f = rate at full stick deflection
* p = rate at partial stick deflection
* x = fraction of stick deflection that generates rate p

Where the input region from 0..x is quasi-linear (as is the output region 0..p).

Then it's trivial to calculate m = log(0.01) / log (x) to know a good exponent, and a = ((p/f)-x) / (x^m - x) to know a good exponent fraction.

https://www.desmos.com/calculator/xtdd08smbk

Three examples.  Red is where half stick deflection is asked to make 10% control rate.  Green is where 75% stick deflection makes 60% control rate.  Blue is where 10% stick deflection makes 5% rate.

mlyle, 2015-08-11:
For purposes of load to reverse this in gcs... 

x = e^(log 0.01)/m
p = ( a x^m - ax + x ) * f

Trex4Git, 2015-08-11:
That is a nice idea. 
I will give the user work control on the behavior, and a tuning more similar to MWRate or Acro+ would be possible with the advantage of still PID control at the "high end". Maybe Acro+ is than not needed anymore.
On IRC I have seen that maybe I should implement this ;) 

So, can promise how fast that will be done, as i have 2 PR where some work ins needed.
But this may be not much work.

As a start just a quick plot, similar to the expo plots i have done on the acro thread on forums:

[image]

The load for the µC might be higher than expo3 function, but I think it should my not much more cycles.

peabody124, 2015-08-11:
I think for practical purposes keeping the number of adjustable parameters minimal is really important. Otherwise in my experience people will just keep keep hunting and get overwhelmed.

I'm also really curious (i hesitate to use the word skeptical) how much someone would really want more degrees of freedom than a curve that looks like x^3 with the max rates adjustable. Again, I'm happy to change my mind from people's flight tests since this stuff is really all "feel".

Also don't confound this with things like gyro(pid) suppression, that creates different dynamics and can't be mapped to an equivalent rate for a stick position. Specifically if you were to have a step that goes from requesting 0 deg/s to 500 deg/s in rate mode, you would get a significantly different transient response than if you were to feed a step into ```mwrate/acro+``` where you have a static offset to the outputs and the feedback controller attenuated.

mlyle, 2015-08-11:
> I think for practical purposes keeping the number of adjustable parameters minimal is really important. Otherwise in my experience people will just keep keep hunting and get overwhelmed.

So we have the nice UI to display rate curves.  Basically I want to remove the "exponential %" from it ultimately, and let you move around a point for the linear region.   So you can say, 100% deflection is 1280 degrees per second, and linear to 75% deflection which is 360 degrees per second, and it'll find the appropriate exponent.

> I'm also really curious (i hesitate to use the word skeptical) how much someone would really want more degrees of freedom than a curve that looks like x^3 with the max rates adjustable.

I want an exponent higher than 3.  I want something linear at low rates.

> Specifically if you were to have a step that goes from requesting 0 deg/s to 500 deg/s in rate mode, you would get a significantly different transient response than if you were to feed a step into mwrate/acro+ where you have a static offset to the outputs and the feedback controller attenuated.

Well, of course.  The acceleration is a little different-- snap and then underdamps to the new equilibrium rate.  But there's benefits to keeping control and at the highest rates it may be hard to tell the difference.

mlyle, 2015-08-11:
Current exponent of 3 is "linear" to 21% of scale per current math.

Trex4Git, 2015-08-11:
Just another plot, to show the behavior that can be achieved a little bit extreme.
Almost everything the same as above, but just moved the "linear cutoff" point for stick input from 0.6 to 0.9:

[image]

mlyle, 2015-08-11:
https://www.desmos.com/calculator/okxto0ufnd   makes it easy to play with by moving sliders.

fujin, 2015-08-21:
> icee> OK I suggest.. 1600 / 85% / exponent 7

Trex4Git, 2015-08-23:
Is it planed also to move the Horizon Expo the ExpoM, or only the Rate and Attitude ?

Because for Horizon Expo I implemented in GCS plotting markers for the transition from attitude to rate. Therefore I needed an inverse  expo3 function. 
So for ExpoM than also an inverse function will be needed, or we drop plotting the markers.

mlyle, 2015-08-23:
Yes, @fujin mentioned that on irc.. I have a to-do to make an inverse function.. But first some early input from flight test will decide whether we try and get this into this release.