TauLabs/TauLabs #1786: Maybe a strange priority of "low level" tasks for flight control

Trex4Git, 2015-08-11:
I noted maybe a strange priority of the "low level" tasks for flight control.
The MPU task and stabilization task have highest priority, but sensor task and attitude task have "only" high priority.

This is how it seems to me:
- The data gets read on MPU task and is send than to sensor task with a queue. 
- Than in sensor task some data processing is done, and the final data, e.g. gyro are stored in a UAVObject. 
- The stabilization task gets than "triggered" by the gyro UAVObject update event.

So when we want a fast processing with low delay/jitter, the sensor task (as beeing a part of the data flow chain) also needs highest priority, i think.

Also some flightmodes in stabilization task need data from attitude task, and the accel data update is extra done before the gyro data update (https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Sensors/sensors.c#L207-L211). 
Maybe the attitude task should also be highest priority.

peabody124, 2015-08-11:
No this is intentional, it sets up a priority chain of updates so the subsequent modules consume that data with minimal latency.

closed 2015-08-11 by peabody124

peabody124, 2015-08-11:
Sorry I hope closing that didn't seem dismissive, but you don't want sensor to be the highest priority because things like actuator, stabilization should be higher and preempt it.

Trex4Git, 2015-08-11:
No, it doesn't seems dismissive to me.
I just thought it was strange, and last times there were some jitter problems, so i wanted to discuss that.

I have not much experience with RTOS yet,and not a complete overview of tasks, priorities, and dependencies in taulabs.

I can understand why that sensor task has not the highest priority.
But can't this  be a possible reason of jitter? 

Because to MPU task, that gets triggered from the MPU IRQ has highest priority.
It reads the data, does some processing and than puts the data in the queue, from what the sensor task gets triggered. 
But as not having the highest priority it can be preempted by tasks with highest priority. 
And stabilization task gets not triggered until sensor task has run, and the data in the uavobject is updated. 
So it doesn't matter if stabilization task has highest priority, if sensor task gets delayed, also our PID control in stabilization task gets delayed.

And why is in attitude task extra mentioned that
```
// Update gyros after the accels since the rest of the code expects
// the accels to be available first
```
https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Sensors/sensors.c#L207-L211

Attitude task /stabilization task seems both get triggered when there is new data in the gyro & accel UAVObjects / gyro uavobject
- attitude: https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Attitude/attitude.c#L413-L414 / 
- stabilization: https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Attitude/attitude.c#L413-L414
 - it says that it waits for the attitude object is updated
 - but here the gyro object is connected to that queue and connecting the attitude object is commented  https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Attitude/attitude.c#L413-L414
So sensor task updates accels and gyros, and the attitude task can not preempt, because it has same priority. But the stabilization task can preempt, and as the gyro object is updated he will run with old attitude data. And some flightmodes need attitude data.

But if everything I have written here is just a wrong understanding how the things should work, than this can keep closed.

mlyle, 2015-08-14:
>And stabilization task gets not triggered until sensor task has run, and the data in the uavobject is updated. 
>So it doesn't matter if stabilization task has highest priority, if sensor task gets delayed, also our PID control in stabilization task gets delayed.

This is exactly right, and the reason for the priorities being the way they are.  If we have fallen behind a small amount-- and another sample is showing up-- it's better to finish sending our output than to consume the sensor value and fall further behind.

It also lets us successfully cope with the sensor rate being too high for the CPU.  If sensor is set higher than the CPU can handle, actuator being higher priority will at least ensure that we send out some control output.

While this in particular looks correct, I do think we should audit all of the priorities to make sure they still make sense at some point in the future.