TauLabs/TauLabs #1789: Makefile - Tools

gke, 2015-08-12:
It would be worth updating the Makefile to import all of the tools required for compilation as it once did and as OP still does. This would make it a little easier for others to contribute to the coding efforts. We seemed to have degraded to an ad hoc process which is marked as a WIP.

tracernz, 2015-08-12:
It does? Here's a snippet of the make output help target (the default target):
```
[Tool Installers]
     qt_sdk_install       - Install the Qt tools
     arm_sdk_install      - Install the GNU ARM gcc toolchain
     openocd_install      - Install the OpenOCD SWD/JTAG daemon
        $OPENOCD_FTDI     - Set to no in order not to install legacy FTDI support for OpenOCD.
     stm32flash_install   - Install the stm32flash tool for unbricking boards
     dfuutil_install      - Install the dfu-util tool for unbricking F4-based boards
     android_sdk_install  - Install the Android SDK tools
     gui_install          - Install the make gui tool
     gtest_install        - Install the google unit test suite
     astyle_install       - Install the astyle code formatter
     openssl_install      - Install the openssl libraries on windows machines
```
QT SDK and ARM SDK are the only requirements to build flight code and GCS. There are platform specific dependencies which aren't included anywhere except the wiki. Maintaining those in the Makefile would get a bit messy.

The Windows environment is a bit of a pain to setup and could do with some work. Perhaps that's what you were meaning with this issue?

gke, 2015-08-12:
Thanks - doing this now. I tried to install all of the tools as set out in the Makefile. Some of the links no longer work with the odd 404. I then found this page which further served to confuse. Why two different processes rather than one clean one like make install_tools.

My reason for getting into this is to get auto-tune working on the original CC board that came to me directly from Dave D.

I will leave this open for now.

Cheers

mlyle, 2015-08-13:
I'm not sure autotune will work on original CC.  We keep it working on CC3D, but CC uses a bunch more ram for the ADC queue.