TauLabs/TauLabs #1792: When sanity checks fail we should panic.

mlyle, 2015-08-14:
Through the code I see a lot of things like:

        if (PIOS_RFM22B_Validate(rfm22b_dev)) {
                return rfm22b_dev->deviceID;
        }

If we have a garbage pointer, we should stop running.  Something has badly gone wrong and we shouldn't continue to execute flight code with undefined results.   Comments?

peabody124, 2015-08-14:
Panic in the sense of put a high priority task into a tight spin loop and block the rest of the RTOS? I think in a debug compile that is appropriate. For a release compile, we should try and keep the system up as much as possible.

That being said when basic assumptions like this fail, who knows how long we are going to keep running. But planning for essentially segfault is maybe not the best option.

mlyle, 2015-08-14:
I think we should reset, stop motors, etc.  It's important to fail as fast as possible in a deterministic way instead of trying to go forward.

When I've worked on far less safety critical stuff in the past, this has always been teh

Normal failsafe, we at least know the basic situation-- we've lost the receiver or the stuff from the receiver doesn't make sense anymore.  Here, we've lost the receiver and unknown things have happened to our system state.

If pointers point to other stuff, random things have been written to sections of memory, or power supplies have sagged and we've lost some RAM contents, and any variable we touch could be tainted.  We could be acting on arbitrary things, sending pulses to the wrong place, who knows what..