TauLabs/TauLabs #1796: OpenLRS problem when using non-default number of channels

OhmEye, 2015-08-14:
Tested with release and next, and OpenLRSng 3.8.5 on the TX. Setting the number of channels to 12 on the TX using the OpenLRGng Configurator results in sporadic motor cutouts apparently randomly a few times per minute. For a visual example see https://www.youtube.com/watch?v=8Agt80uOLPo

I have not tested other than 12 channels, and have tested these TX settings:

default
default + telemetry off
default + telemetry off + data rate 9600
default + telemetry off + data rate 9600 + channels=12
default + channels=12

I have not observed motor cutouts when channels are set to 6, but the cutouts always occur when channels are set to 12. It appears to be manifesting sporadic brief failsafe-like events, channel input in GCS can sometimes be seen to be very briefly lost at the time of a motor cutout.

peabody124, 2015-08-14:
Awesome, thanks for repeating this and narrowing it down. I'll try and take a look ASAP. I hope this won't be a nasty issue related to the relative timing on the two targets, but I fear it will be.

mlyle, 2015-08-15:
For clarity, channels here means the number of frequencies used for hopping.

mlyle, 2015-08-15:
Added for consideration for next release, though I'm a little doubtful.

phobos-, 2015-08-15:
Confirmed to happen in these configurations as well every ~25sec:
Datarate 19200:
4channels + 4 switches
8channels
Datarate 57600 (experimental):
4channels + 4 switches
8channels
occasional GCS freezes
Datarate 125000 (experimental, my go to rate):
freezes the GCS permanently after couple of seconds, so unable to check.
CPU load is 19%, telemetry set at 115200bps.

OhmEye, 2015-09-04:
After spending more time testing this, the problem does occur with 6 hopping freqs, just apparently less often. Here is an example of a cutout even happening during flight with default TX settings:

https://www.youtube.com/watch?v=ImLIX16F8_w

peabody124, 2015-10-05:
I think I have fixed this. Here is a link to a test firmware (based on next) that I'd love to have feedback on https://drive.google.com/file/d/0ByPItuei3Xt6RDk5Z2pFNnlCQVU/view?usp=sharing

acceleroto, 2015-10-06:
I just loaded the new firmware and did a couple quick test flight inside since it's raining cats & dogs outside. No issues so far, but I'll test more when the weather clears.

One interesting thing that happened with the previous firmware was any time I selected autotune, the quad would auto-crash instead. I checked & double checked all the settings, but I couldn't get it to enter autotune. I'm not seeing the auto-crash any more. I didn't change any settings when I loaded the new firmware, so perhaps that's related to the OpenLRS problems.

More as soon as I can...

OhmEye, 2015-10-06:
FWIW, I never had any issues with autotune other than lasting the duration without a motor dropout. I have not tested this latest fix yet.

OhmEye, 2015-10-06:
@peabody124 do you have a matching GCS version I can grab and build? Seems next is unable to save some objects, I get a "Objects not present on the hardware: ActuatorSettings" when trying to save motor output config. Is there a specific next commit # I should be using? Or am I doing something wrong?

mlyle, 2015-10-06:
Would be nice to have code so that I can test on my branch, too..

peabody124, 2015-10-06:
@OhmEye I'll double check when home, but Im pretty sure that was built on next. What operating system do you need? I can also patch it against most recent release.

OhmEye, 2015-10-06:
@peabody124 Linux64. I can try pulling latest next though, mine is 10 days old or something, just wanted to ask before doing another build, it takes a while on this box. :)

OhmEye, 2015-10-16:
@peabody124 I haven't flown it extensively yet but no apparent glitches during the first couple batts. Can we please get your fix into next?

acceleroto, 2015-10-16:
I haven't flown it much either - maybe 3 batteries. I let it sit on the bench with everything turned on for about 30 minutes and didn't see any issues either. I was going to test more this week, but I smoked my TL PDB. It'll be a couple weeks before I have my Tau Labs/Sparky 2 quad back in the air.

OhmEye, 2015-10-26:
I flew for another several batteries over the weekend and saw no issues at all on default olrs TX settings. I haven't tried any non-defaults. Was the only quad I had with me for two meetups so got all my airtime. Looks like this is merged into next now, yay!

dustin, 2015-11-26:
If this is believed fixed and nobody is working on it, it might be worth closing.  There are some changes @peabody124 has applied which are reported to have increased robustness significantly.  I've flown it on my 150 and that's not something I've had problems with.

peabody124, 2015-11-26:
Good suggestion

closed 2015-11-26 by peabody124