TauLabs/TauLabs #1801: PicoC disabled on RevoMini

mlyle, 2015-08-17:
commit dd588113360d28cf35e8a4d554022ff95bbfd822
Author: ernie of infect <ernie@web.de>
Date:   Fri Sep 5 07:19:41 2014 +0200

    picoC: remove revomini support
    
    Temporary removed due a compiler issue on make ef_. Will be enabled
    again when fault is removed.

mlyle, 2015-08-17:
```
 PADDING   [ef|rm  ]  build/ef_revomini/ef_revomini.fw_pre.pad
dd if=/dev/zero count=98304 bs=1 2> /dev/null | LC_CTYPE=c tr '\000' '\377' > /Users/mlyle/taulabs/build/ef_revomini/ef_revomini.fw_pre.pad && [ ${PIPESTATUS[0]} -eq "0" ]
 PADDING   [ef|rm  ]  build/ef_revomini/ef_revomini.fw_post.pad
dd if=/dev/zero count=-17688 bs=1 2> /dev/null | LC_CTYPE=c tr '\000' '\377' > /Users/mlyle/taulabs/build/ef_revomini/ef_revomini.fw_post.pad && [ ${PIPESTATUS[0]} -eq "0" ]
```

mlyle, 2015-08-17:
```
FW_POST_PAD := $(shell echo $$[$(FW_BANK_SIZE)-$(FW_DESC_SIZE)-$(FW_DESC_SIZE)-$(shell wc -c < $(FW_BIN))])

FW_BANK_SIZE        := 0x00040000  # Should include FW_DESC_SIZE (256kb)
FW_DESC_SIZE        := 0x00000064
-rwxr-xr-x  1 mlyle  staff  279632 Aug 17 02:24 fw_revomini.bin
```

274k vs 256k.  384k on part total.  Bootloader occupies first 32k, next 96k unoccupied I think, then 256k of flash.

Without picoc we are at 216k.   So picoc increases our image size by a startling 58k.

It may be time to claw back some of that 96k.  We're already building with -Os.  Or maybe we give up on picoc on revomini.

ernieift, 2015-08-19:
It think it is a linker problem not a compiler error. Have a look at this https://github.com/TauLabs/TauLabs/commit/290690f58f1a18e52fc004d287591da655f6eda6 . It should solve the issue on revomini. I don't have it, so i can't test.

tracernz, 2015-08-19:
The problem is that there isn't enough space available in the firmware section to fit it. :wink: The linker script we're using for all F4 targets has 896 k in the FLASH section which is correct for the 1 mb part but not the 384 k part used on the revomini, and so you don't get an actual linker error. If the linker script were correct you'd get something like this when you try to link the elf:
```
arm-none-eabi/bin/ld: region `FLASH' overflowed by 14392 bytes
```

ernieift, 2015-08-19:
Why Quanton and Sparky2 work and Revomini don't?

tracernz, 2015-08-19:
I've just been told the Revo mini is actually a 1 mb part after all? I guess we just need to use more of that for firmware.

ernieift, 2015-08-19:
This is what the https://github.com/TauLabs/TauLabs/commit/290690f58f1a18e52fc004d287591da655f6eda6 do. We had the issue first on quanton. sparky2 was developed without this issue. Only the revomini is been forgotten, i guess.

tracernz, 2015-08-19:
Yeah, exactly. :) They all have the same MCU.

mlyle, 2015-08-19:
@ernieift My bad, I thought the actual end of flash was a real end of flash and not something where we were saving sectors.

I sent a pull request, but won't be able to test until my revomini-clone shows up from China.

mlyle, 2015-08-24:
Confirmed to work.

closed 2015-08-24 by mlyle