TauLabs/TauLabs #1807: Gyro cutoff frequency change problematic

mlyle, 2015-08-18:
@Trex4Git Please confirm my reasoning here.

@pug398 I believe this is the issue you've seen.

It looks like the changes to GyroCutoff in PR #1765 have detuned existing Quanton setups.  I believe the old Sparky2 default cutoff was 41Hz.  If my reasoning is correct, the old Quanton cutoff for the Gyro LPF was 55Hz with default settings (@Trex4Git is this true?).  We picked a compromise value of 45Hz.  Unfortunately, this means Quanton has more filter delay than it used to, and existing autotunes have too much PID gain as a result.

On one level, this is good news-- it means that autotune is getting fairly optimal values for the PID loop and a small introduction of additional delay results in an overtuned situation.

I can think of two possible approaches to fixing this.

1. Setting our default Gyro LPF to 55Hz on all platforms.  I think this value is safe from a vibration perspective on most vehicles, and reducing delay is always safe-- bonus phase margin.  Quanton would continue to operate the same and people with Sparkies would pick up some more PID headroom.
2. Document that retuning is necessary.

Any thoughts?

Trex4Git, 2015-08-18:
I based the default value on 500Hz MPU rate, what seemed to me the default setting for most targets.
Quanton has 666.67Hz so had a higher cutoff with the old implementation (where cutoff changes with MPU rate).

When i remember correctly about my investigation about the cutoff frequency it was with default GyroTau (0.005):
 500Hz:  40.5Hz
 666Hz:  54Hz

So 55Hz should be ok, in cleanflight a lot of people seems to use 60Hz at the moment.

peabody124, 2015-08-18:
has anyone plopped in debug output to see the calculated gyro_alpha that #1765 produces (compared to the original 0.005), and also to make sure it is nice and stable?

i'm also putting some comments in the original PR that should be fixed with the next fix. That PR was marked WIP so I never had a chance to review it.

peabody124, 2015-08-18:
Yeah option 1 seems ideal. And nice thing is people can test without a new build by changing the param :)

mlyle, 2015-08-18:
So old gyroalpha was sqrt(1/e) = 0.607.

Quanton/Colibri/AQ32/FlyingF4 (667Hz):
New value @55Hz, assuming 667Hz update rate is: e^(-2 * pi * 55 * (1 / 667)) =~ **0.595**
New value @45Hz, assuming 667Hz update rate is: e^(-2 * pi * 45* (1 / 667)) =~ 0.654

Sparky2/CC3D (500Hz):
New value @55Hz, assuming 500Hz update rate is: e^(-2 * pi * 55 * (1 / 500)) =~ 0.500
New value @45Hz, assuming 500Hz update rate is: e^(-2 * pi * 45 * (1 / 500)) =~ 0.568
New value @40Hz, assuming 500Hz update rate is: e^(-2 * pi * 40 * (1 / 500)) =~ **0.604**

Sparky1 (444Hz):
New value @55Hz, assuming 500Hz update rate is: e^(-2 * pi * 55 * (1 / 444)) =~ 0.459
New value @45Hz, assuming 500Hz update rate is: e^(-2 * pi * 45 * (1 / 444)) =~ 0.529
New value @40Hz, assuming 500Hz update rate is: e^(-2 * pi * 40 * (1 / 444)) =~ 0.568
New value @35Hz, assuming 500Hz update rate is: e^(-2 * pi * 35 * (1 / 444)) =~ **0.609**

FlyingF3 (380Hz)
New value @31Hz, assuming 500Hz update rate is: e^(-2 * pi * 31 * (1 / 380)) =~ **0.599**

Trex4Git, 2015-08-18:
I have not done hardware debugging for https://github.com/TauLabs/TauLabs/pull/1765 to test the values,
as i have not running it in my development environment.

Just tested the math before in matlab, as you can see in some of the plots.

Answered also to @peabody124 comments on https://github.com/TauLabs/TauLabs/pull/1765.
I can put everything together with this in a new PR.

mlyle, 2015-08-19:
Reverted the gyro cutoff code in #1811.  Something's wrong.  I will read and study in detail later.

mlyle, 2015-08-24:
Flown by @pug398, verified fixed and merged.

closed 2015-08-24 by mlyle