TauLabs/TauLabs #1812: S.Port Telemetry not working correctly with OpenTX 2.1

WZ9V, 2015-08-19:
The current S.Port implementation is causing duplicate entries to be created on the Taranis during discovery when using OpenTX 2.1 causing all telemetry slots to fill and triggering a recurring error.  My understanding is that the Teensy Mavlink converter project has similar issues with OpenTX 2.1

mlyle, 2015-08-20:
@WZ9V:  Can you please try current next?  @xdandys submitted #1731 a few weeks ago for this issue.

WZ9V, 2015-08-20:
Where would I get that one from?  I'm still pretty new to TauLabs.

mlyle, 2015-08-20:
If you need Windows, windows builds were broken for awhile but are now available from: http://jenkins.taulabs.org/artifacts/

Note that the next branch is bleeding-edge and may have critical bugs.  It's development in preparation for the next release.

WZ9V, 2015-08-20:
I downloaded the 8/19 Windows one from Jenkins and it did indeed fix the S.Port telemetry problem with Open TX 2.1.1 but it was a real pain getting a stable radio calibration and vehicle setup gave me manual, stabilized1 and stabilized1 for my three flight modes.  I'll start over again tomorrow and see if it acts weird again but I can at least confirm that the telemetry worked properly.  Thanks.

mlyle, 2015-08-20:
OK.  Vehicle setup doesn't setup flight modes.  I will close this issue as it appears PR #1731 is indeed good.  Thanks for trying it out!

closed 2015-08-20 by mlyle