TauLabs/TauLabs #1821: Sending IAP reset to Sparky2 with USB and Bat connected causes motors to spin and servos to move

mlyle, 2015-08-22:


peabody124, 2015-08-22:
Crap. I thought this code: https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky2/bl/pios_board.c#L53 was sufficient to prevent this but apparently not

Possibly we need to half all timers and reset their outputs to make sure the resources are unallocated. This problem likely applies to all the targets.

Maybe there is another command we can issue in the common code which resets all peripherals to the power state.

peabody124, 2015-08-22:
If you want to try it, perhaps add these lines  before L53 in the boot loader (and of course make sure to flash a new boot loader):

```
	TIM_DeInit(TIM1);
	TIM_DeInit(TIM2);
	TIM_DeInit(TIM3);
	TIM_DeInit(TIM4);
	TIM_DeInit(TIM5);
	TIM_DeInit(TIM6);
	TIM_DeInit(TIM7);
	TIM_DeInit(TIM8);
	TIM_DeInit(TIM9);
	TIM_DeInit(TIM10);
	TIM_DeInit(TIM11);
	TIM_DeInit(TIM12);
	GPIO_DeInit(GPIOA);
	GPIO_DeInit(GPIOB);
	GPIO_DeInit(GPIOC);
	GPIO_DeInit(GPIOD);
	GPIO_DeInit(GPIOE);
	GPIO_DeInit(GPIOF);
```

mlyle, 2015-08-27:
So @tracernz and I investigated.  and didn't really come to a coherent set of ideas.

It may be best, when we're in the wait-for-IAP-reset 500ms state to stop PWM.  This would let ESCs disarm.

But there's other reset scenarios-- brownout, watchdog, etc.  Reset creates things that look like PWM pulses.  This all takes thought.

mlyle, 2015-08-31:
So, we don't really know what to do and ... I think we need to be systematic.  Probably the best mechanism (suggested by @tracernz) would be sure that all outputs are pulled low during the window we're waiting for IAP reset so ESCs can disarm.

This is an unfortunate problem, but I am having trouble seeing a path beyond deferring it out of this release pending more thought and consideration.

mlyle, 2015-08-31:
Also note that pulling output low during the wait for IAP reset (500ms) fixes this specific case but doesn't address resets from bad power, watchdog, etc.  It does seem the bootloader gets the pins low relatively quickly.

One thing that exacerbates the ISP case is that with many configurations the reset happens just after the PWM pulse is driven high.  So it's driven high hard, and then is weakly pulled up during reset.

tracernz, 2015-08-31:
>So, we don't really know what to do and ... I think we need to be systematic. Probably the best mechanism (suggested by @tracernz) would be sure that all outputs are pulled low during the window we're waiting for IAP reset so ESCs can disarm.
This adds an additional layer of safety to normal/controlled resets which is certainly welcome.

I still can't think of a foolproof way to handle failure resets. I assume the the peripherals will be in normal reset state after e.g. a watchdog reset? If that is the case the solution proposed by @peabody124 would be ineffective at changing behaviour? The reset state is tri-state with pullup disabled according the F4 datasheet. That doesn't really explain why the pins are pulling high during reset... Need to look into this a little more..

It's worth noting that the [Sparky2 bootloader code to set the PWMs as low outputs](https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky2/bl/pios_board.c#L53) is specific to that target, at least I don't see it on the Quanton. It also doesn't actually set them as low outputs, but rather as alternate function?

peabody124, 2015-08-31:
I'm not totally sure the peripherals _are_ reset after powering up, TBH. Otherwise I'm not sure where there would have been a long enough delay before hitting the chunk of code that zeros them out would be.

tracernz, 2015-08-31:
There's about 36 us between the micro putting the GPIO into reset state and that code running according to my scope captures. Not very long.

kubark42, 2015-09-10:
A cursory search makes it look like https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky2/fw/pios_board.c#L257 seems to be unique to Sparky2. Not saying it's related, but it does seem to be a bit of code which touches on this subject and isn't shared by other targets.

mlyle, 2015-09-10:
@kubark42 So, @tracernz, simonk and I investigated a bit on IRC.

It looks like pretty much all targets end up doing something that's unsafe here in one way or another and that we can't make it unconditionally safe.  So the question becomes "what behavior is least broken"?

Hardware reset makes things float high.  If we're controlling the reset-- e.g. IAP, we can drive the lines low for long enough to force a disarm of most ESCs, but that doesn't affect other sources of reset (power glitch, watchdog, etc).  With IAP, the hardware reset ended up being synchronous to the PWM pulse most of the time on Sparky so that makes it worse (driven strongly high then left weakly pulled high).  But no matter what this is going to produce something that could look like a pulse.