TauLabs/TauLabs #1841: "Large" UAVObject size causes crash on sparky2

ekaszubski, 2015-08-28:
Summary:
- If the total size of any UAVObject is greater than 244 bytes, sparky2 goes into a reboot loop

Platform info:
- Core-i7 CPU
- ubuntu 12.04.5 amd64 / gcc-4.8.1
- sparky2 connected via USB (settings wiped)

Software
- branch: next (8f5764990d99192d1abf0cde28904b86a5a54ae6)

Issue Details:
- UAVObjects size is initially 3942
- StabilizationSettings size is initially 222 bytes
- modified shared/uavobjects/stabilizationsettings.xml; added 23 bytes of data via one 16-element uint8 vector and one 7-element uint8 vector
- compilation verifies that UAVObjects size is now 3965, StabilizationSettings is now 245 bytes
- flash firmware; result: reboot loop, with "Telemetry disconnected" each loop

Resolution Details
- reduce 7-element vector by one element; compilation verifies that UAVObjects size is now 3964, StabilizationSettings is now 244 bytes
- flash firmware; result: sparky2 functions as expected (boot, connect to USB telemetry)

Additional Details
- attempted to break ManualControlSettings in the same way
- ManualControlSettings is initially 112 bytes
- added eight 16-element uint8 vectors and one 5-element uint8 vector
- compilation verifies that ManualControlSettings is now 245 bytes
- flash, boot loop
- reduce 5-element vector by one element; size is 244; flash, no issues

I'm guessing this is a result of the message size being stored as a uint8 somewhere, but the message size has some additional length added due to headers, etc, so going above 244 causes an overflow or some other unexpected behavior.

kubark42, 2015-08-28:
You're correct. The buffer is only 256 bytes big. This is relatively easy to see in the case of FAST_HEAP at https://github.com/TauLabs/TauLabs/blob/next/flight/UAVObjects/uavobjectmanager.c#L761

The StabilzationSettings is one of those "everything and the kitchen sink" UAVOs. There are good arguments for making some kind of new controls parameter category and splitting StabilizationSettings into smaller chunks. Just throwing ideas out, maybe one including saturation limits and another tuning parameters.

mlyle, 2015-08-28:
@kubark42 I'd like to make flight modes much more general.  I'm inclined to say we should have a multi-instance object for this along with something like a union in uavo, so we can have mode-specific parameters.  Would let us have two different rate mode tunings in place at different flight mode switch positions.  It would also save memory since we wouldn't need to have all the data for modes we're not using.

It probably wouldn't quite encompass everything in stabilizationsettings though.

mlyle, 2015-08-28:
Probably should teach uavobjectgenerator to reject big ones.

ekaszubski, 2015-08-28:
@kubark42 Many thanks for confirming. At the very least, I feel like this limit should be well-documented. The changes I was making weren't that out-of-the-ordinary; it only takes a few extra float vectors to hit the size limit in an already-large uavobject.

@mlyle Agreed, until a more general solution is decided on, I feel like the generator should at least throw an error in these cases. When sparky2 started crashing, this issue was the last thing I would have expected to be the cause and it was a pain to track down.

kubark42, 2015-08-29:
@mlyle, you mean like the SystemSettings UAVOs? Perhaps there's a better way of doing it than was done for SystemSettings, but that UAVO seems fragile and likely to produce obscure bugs. It's memory efficient at the cost of readability. My preference is to make UAVOs more atomic in scope (this also has the nice side effect of not nuking settings, e.g. tx channel settings when a new flight mode is added).

> I'm inclined to say we should have a multi-instance object...

Multi-instance would help here (and in the MixerSettings UAVO as well). Unfortunately, settings UAVObjects don't support multi-instances, so they would have to be extended. 

A few months back @BrainFPV and I looked and it seems this should be possible, but we never dove into it to see if there were any gotchas.

mlyle, 2015-08-29:
@kubark42 Yah, I mean adding something like explicit unions of structs to uavo, with a selector for which applies.  So the selector would be flight-mode, and there'd be a flight-mode specific configuration struct that would follow.

And yah, I'm in favor of settings multi-instance objects.. then the whole need for waypoint partitions etc goes away.

kubark42, 2015-08-29:
It's worth discussing to see what the advantages/disadvantages are. RAM usage is clear, but I'm wondering if there are other subtle differences between making several small UAVObjects grouped under a new category vs. a union.

+1 on resolving the need for waypoint partitions, etc... We were looking at multi-instance settings for optical flow sensors where we have one UAV setup that has something like 12 cameras.

PR #1849 (2015-08-29): UAVOGen: Add max UAVO size limit
Fixes #1841. I think my overhead math is correct based on what is stored and it certainly lines up with #1841 which I was able to reproduce and confirm this fix with.
closed 2015-08-29 by mlyle