TauLabs/TauLabs #1850: Collective pitch functionality appears broken

ekaszubski, 2015-08-28:
Hardware
- 480-class collective-pitch helicopter, 120 degree swash
- sparky2
- DSMX satellite receiver (SPM9645)
- Walkera devo10 with deviation firmware in 9-channel DSMX mode

Issue
- My transmitter is set up to output full PWM range [-100,100] for collective, as with all other channels
- On the input menu in GCS, I have the collective neutral set to the zero-stick position, which is approximately half-way between the min and the max PWM
- Vehicle type is set to helicopter, 120 degree CCPM, collective pass-through
- Collective curve in GCS is 1-to-1 (or more specifically, [-1,1])
- For positive collective stick values, the swashplate works as expected
- For negative collective stick values, the swashplate snaps to approximately zero collective

Partial solution 1
- in GCS, under input, set neutral point of collective to min collective channel value
- now, collective works as expected, but when the transmitter is shut off, the swashplate snaps to full negative pitch, which can cause catastrophic results in-flight

Partial solution 2
- based on the MixerCurve function implementation (Actuator/actuator.c), setting the first point in the collective curve to a value <=-1 will cause the value to be passed through as-is
- it appears this must be done manually; not sure what effect the "collective passthrough" option under vehicle setup actually has; maybe it was supposed to do this?
- this works as a pass-through, but causes the remaining points in the collective curve to be ignored

Full solution
- Upon further inspection, it appears that the MixerCurve function (Actuator/actuator.c) expects its input to be on the rage [0,1], which make sense for throttle curves; however, the collective curve *should* be defined on [-1,1], should be configurable, and should default to a position of 0 (or some other configurable value) when the transmitter is not connected
- MixerCurve should be renamed to ThrottleCurve; it should continue to handle input in [0,1]
- A new function, CollectiveCurve, should be created to handle input in [-1,1]
- ThrottleCurve should be replaced with CollectiveCurve for MIXERSETTINGS_CURVE2SOURCE_COLLECTIVE
- I have this solution implemented and extensively tested in a separate branch

tracernz, 2015-08-28:
Nice catch. I think our heli support is very lightly used (and I haven't yet tried it myself) so it's nice that someone is actually testing it. Feel free to submit a PR with your fix, it sounds good to me. :+1:

mlyle, 2015-08-28:
Please open a PR on this one. :)

mlyle, 2015-08-31:
Fixed #1859.

closed 2015-08-31 by mlyle