TauLabs/TauLabs #1854: Very long command lines on link can fail (noticed on Windows)

mlyle, 2015-08-28:
Right now we pass fully qualified include paths over and over and a very large set of defines about what UAVOs to include on the target.  We do this even for linking where -I and -D parameters aren't used.

This can result in a command line too long for process spawn.

Reported by @DTFUHF

mlyle, 2015-08-28:
This happened on flight infrastructure compiling sparky2.  Exacerbated because @DTFUHF had the tree in a relatively deep location.  Still, we should do better with this.

kubark42, 2015-08-29:
Our jenkins path added 20k characters to the 17k characters for the ```CreateProcess``` command. Our solution was to map the root directory to a virtual drive in Windows. Not claiming this is the best approach in general, but it worked quickly and easily in our case.

mlyle, 2015-08-29:
The low hanging fruit i see is getting the -DUAVOBJ ... stuff out of the command line and into an auto-generated header.

Also can get the include paths and defines out of link command lines, which are the longest command lines.

Eventually we should be forming libraries rather than trying to keep track of all the objects everywhere.