TauLabs/TauLabs #1862: Stack overflow on FlyingF4

ducminhkt, 2015-08-29:
I just update firmware build on 27/8/15 (367e90b).  I have a problem when config my quad. The system health show Stack Overflow error when I turn on TX. See images below
[image]

Stack remaining: Before turn on TX
[image]
Stack remaining: After turn on TX
[image]
System Alarm
[image]
System Stat
[image]
My hardware
[image]
Input config
[image]
Task running
[image]

In Task remaining, you can see the value at Manual Control come to zero (some time < 60) when I turn on TX. If I boot up my Quad when turn off TX, Stack Overflow is OK. 
Before update this firmware, I use version 744c79e  (8 days ago), Stack Overflow is OK.
Please show me where's my mistake, or because firmware bug ?

mlyle, 2015-08-29:
merges in this range:

    Merge pull request #1797 from mlyle/mpl-mavlinkstackdiet
    Merge pull request #1808 from mlyle/mpl-makededup
    Merge pull request #1809 from mlyle/mpl-deleteoldtrim
    Merge pull request #1814 from mlyle/mpl-removeoplinksettings
    Merge pull request #1815 from mlyle/mpl-gyrofiltahz
    Merge pull request #1820 from mlyle/mpl-loiteraltitude
    Merge pull request #1823 from mlyle/mpl-fixdeadbands
    Merge pull request #1825 from mlyle/mpl-dumbfix
    Merge pull request #1810 from mlyle/mpl-picoconrevo
    ~~Merge pull request #1822 from BrainFPV/enh_use_fastheap_2~~
    Merge pull request #1828 from mlyle/mpl-halexportconfigurecom
    Merge pull request #1818 from mlyle/mpl-1505
    Merge pull request #1824 from mlyle/mpl-toolswindows
    Merge pull request #1791 from peabody124/taulinkmodule
    Merge pull request #1834 from cyphyworks/fix_metadata_multiinstance_initialization
    Merge pull request #1832 from cyphyworks/nan_fixes
    Merge pull request #1833 from cyphyworks/settings_enhancements
    Merge pull request #1829 from mlyle/mpl-nazeserial
    Merge pull request #1835 from cyphyworks/fix_map_NED_mouse_pointer
    **Merge pull request #1830 from mlyle/mpl-1806nodupports** (touches PiOS)
    **Merge pull request #1816 from fujin/acro-plus**  (touches manualcontrol)
    ~~Merge pull request #1855 from TauLabs/revert-1822-enh_use_fastheap_2~~

mlyle, 2015-08-30:
Can you please try pr-1866?   Either you can build or find a pre-built copy at http://jenkins.taulabs.org/artifacts/   (pick the top pr-1866 build for your platform).  I found a problem with a NaN fix that ended up consuming a lot more stack space.

lolorc, 2015-08-30:
oh this looks like what I'm suffering from... unarmed quanton crashes immediatly. not sure it was crashing   while flying yesterday. gonna give pr-1866 a try

lolorc, 2015-08-30:
116bytes AH stackremaing without #1832. (i've reseted to board several time, i also got 140bytes for few seconds and then goes down to 116bytes)
once enabled there're 20bytes left.
I'll fly test it this afternoon.

mlyle, 2015-08-30:
@lolorc If you want to follow next, you have to accept there's a chance for breakage when really unintuitive things happen.  Read the PR.  It is nowhere near AH in the code.  It also didn't happen on all targets.  A flight test program involving altitude hold across many targets for each PR sequentially is a rather substantial undertaking.

Thank you for the confirmation that AH works with #1832 reverted and that #1832 increased stack consumption in the altitude hold task. 

I think it's hilarious you're complaining about insufficient test on a volunteer project and are scared of a PR so you're not going to test it.

mlyle, 2015-08-30:
@ducminhkt #1866 has merged.  Could you please try and see if this issue is resolved?

ducminhkt, 2015-08-31:
@mlyle: thanks, I will test it tomorrow.

lolorc, 2015-08-31:
@mlyle oh you're so wrong, I'm not complaining about lack of tests, I'm complaining about hastily broken merges. I've been testing and running next for a while, I know exactly what to expect thank you very much.
Still it's been broken every day or so lately, you're right this is a community-driven project, you don't leave users time to test PR and the associated branches, you just decide to merge too quickly imho.

When you comment with "Still kinda a mystery how the original PR affects stack usage in this particular task. Oh well￼" I don't find it hilarious at all, trust level just goes down...

"bench test" and "flight test" should probably get defined with proper lists of checks to validate before merging potentially critical stuff...

tracernz, 2015-08-31:
@lolorc, while I can appreciate what you're saying you do need to realise that the release strategy is a little different this time. The idea is to get everything in much earlier so we can have a full month of testing prior to release. Rather than lots and lots of time spent testing everything on separate branches the emphasis is more on testing the entire system (this is not to say that branches aren't tested prior to merge however). @icee is just trying to do his best to please everybody by getting all the features we want in before the window closes.

mlyle, 2015-08-31:
@lolorc Please enumerate what set of tests you think should have happened for the NaN merge.  Go have a look at the code first.  It's a really unexpected effect and I expect to be reading some of the output object code to see what's going on.  In any case, we bounded the regression to that particular change using bisection and were able to confirm that the stack usage increase was removed by altering the structure of that change.

Stop complaining or start helping out by actually doing some test and validation.  There's multiple times I've kept PRs open for test / comment for weeks with no one pitching in.  

@Trex4Git has had a PR open for 22 days for feedback and test and no one has pitched in.  @guilhermito has had a UI PR open for 14 days with nothing but some cursory test from me.  There are multiple helicopter PRs that would be really nice to get in for this release and that I'm going to bench test but it'd be nice to have other eyes on it.  

It's not uncommon for calls for test to go unanswered for a month.  @pug398 regularly helps out and flies particularly problematic nav code, and @tracernz and others on IRC have pitched in a bunch smoke tested various configurations during the big HAL reshuffle.  (You can see the test matrix that was run for that change https://github.com/TauLabs/TauLabs/wiki/Endomorphism-(September)-release#test-matrix-for-1759 --- which found several configurations that were broken in previous releases.)

I know of 4 times next has broken during this cycle:

1. FastHeapV1.  Was bench tested but crashed some targets on arm.  Reverted.
2. An incorrect commit of my code where a file was missing and some targets wouldn't boot.  Fixed up.
3. FastHeapV2.  May not have been broken / may have been misidentified.  Was thoroughly tested before merge on multiple targets by multiple people.  The crashes were probably actually the NaN fix.  Everyone agrees this should come in as smaller, more-easily verified chunks next time.   Reverted.
4. The NaN fix.   It was trivial, and was bench tested and seen to arm and go through a flew flight modes on revolution for me, and flight tested on Sparky2 as part of the mpl-allstuff branch.  I didn't try altitude hold. No one expected this to greatly increase stack consumption in tasks that were seemingly not altered.  Fixed up.

So we broke 3-4 times.  I'll admit #2 was purely sloppy on my part.  That's not almost every day; we've been at this for a couple of months.

lolorc, 2015-08-31:
yeah I was thinking of that kind of test matrix, could guide people into helping with tests.

`< icee> seriously, fuck lolorc`
`< icee> i hope he dies in a fire`
`< icee> that i set`

kubark42, 2015-08-31:
FWIW, the NaN changes have been flying on internal prototypes for months. It's hard to test beyond the boards you have at hand, and I run all stacks many kbytes larger than necessary (because I don't like fooling around with these reboots) so I wouldn't notice this kind of change. These subtle change can only currently be caught by testing in the wider community.

Without getting into past politics, it's a tough balance. Being the Debian of autopilots just meant that nothing new or bold is tried. I'm unsure how I feel about crashes on arm-- that seems to be a step short of in-air crashes-- but the endomorphism changes are fundamental to the core of the software and so it makes sense to be extra patient during this release. There are bound to be some regressions as the code base morphs so radically, and in general they're caught within hours.

If you're interested in helping out, but testing isn't your thing, I would propose exploring a fund to pay a professional EE to develop a board which would have the ability to do integration testing. Something with an F1, F3, and F4 target, as well as a broad range of sensors and memory chips. This would help us catch real-world problems like this before they get out there into next.

mlyle, 2015-08-31:
This actually could cause in-air crashes, so it's super-bad.

On the other hand, I tested it and never for a moment considered that I might need to test altitude hold on it.  I can't run a full test matrix on everything.

We've broken 3-4 times during this cycle.

* FastHeap V1.  Was tested on some targets, PR was open for 25 days.  @solidgoldbomb correctly spotted how it could break but both of us thought that it didn't apply because of other factors.  No reports of crash before merge.  
* A really stupid break-the-build git mistake by me.  I feel pretty silly.  next could't compile for 12 hours.  This could have been easily prevented.
* This one.  We need tools to watch stacks and do static stack analysis-- that's the only way things like this are going to be prevented.  (Not to mention preventing infrequently-taken-conditionals blowing stack, etc).
* FastHeap V2 was maybe broken.  It was reverted hastily and it may actually have been the NaN change that caused those problems.

lolorc, 2015-08-31:
I can help with tests on quanton, and I have a spare quanton that I can set it up on a frame I care a bit less about.
I live in a town, i can only go to my safe field once a week when the weather permits it...
I guess it's probably not too soon to start doing regular full matrix tests...

mlyle, 2015-08-31:
@ducminhkt sorry about all the unrelated discussion in this bug.  Hopefully your issue was fixed-- please let us know one way or another.

lolorc, 2015-08-31:
with #1832 and #1862, 140 bytes left (and it stays 140bytes with those patches) 52 bytes left when enabled, it doesn't crash. Can probaby try to apply the fastheap v2 on top of that but I was experiencing anything wrong with it few days ago.

ducminhkt, 2015-09-01:
@mlyle Yes, I will test it today afternoon.
I likeTL, not only because it's a good fw, but also it have a developer team with very good support for users. I'm ready to test any new feature from TL.

ducminhkt, 2015-09-01:
@mlyle Tested. Issue fixed Ok. No stack overflow error. Stack remaining of Manual Control always hold at 228 bytes. Thanks.

closed 2015-09-01 by ducminhkt

mlyle, 2015-09-01:
@ducminhkt Thanks for the report and verifying it is fixed.