TauLabs/TauLabs #1875: Remove ERASE_FLASH from targets

mlyle, 2015-08-31:
Would like to remove ERASE_FLASH option because there's another, easier way to do this (right click on settings partition and choose 'erase' in GCS).

(There's an issue open to make this easier #1813).

Also legacy/stale issue #634.

peabody124, 2015-08-31:
Can we keep it as a failsafe? I've still had times when something was totally borked and needed to compile with that flag. I'd hate to end up recreating that code in that case...

guilhermito, 2015-08-31:
The bootloader of F1 targets don't give access to partitions, so we need to keep this compile flag at least for those targets.

mlyle, 2015-08-31:
@peabody124 What circumstances would the right-click-on-settings partition not work (other than F1) but an ERASE_FLASH variant would?

I've been using the ERASE_FLASH variant for awhile, because I didn't know about the GCS settings partition thing.  We have an issue open to make the settings reset easier.

guilhermito, 2015-08-31:
It doesn't work if you are using an older bootloader, but at this stage I believe that in that case most F4 targets are not able of using recent firmwares either due to the change of allocated flash. I believe this wasn't the case for F3's so there could be boards running old bl's.
Whit that said we can always build firmware for all current targets with the flag set and make them into a compressed file available o the wiki or release page somewhere prior to merging this.

mlyle, 2015-08-31:
I favor that-- getting some erase flash firmwares around.  It's always a pain when I've had to deal with it.  I was so happy when @tracernz showed me the erase flash thing in the bootloader.

mlyle, 2015-08-31:
Since this is more contentious and there's version dependencies etc, removing it from the release for further consideration.

AlessioMorale, 2015-09-01:
I've implemented the erase settings from "rescue mode" feature in OP. Instead of making the bl aware of flash configuration I've defined a way to pass commands between bl and firmware (using IAP functionalities) so that firmware can take care of that at startup. This may better fit with the flash size constraints for F1.