TauLabs/TauLabs #1882: Brain FPV GCS Program Crash Windows 10

Icemel, 2015-09-01:
The Tau Lab GCS has been working fine but all of a sudden it has started crashing! I start the program (I'm running Windows 10) and it loads fine, but as soon as I connect the Brain FPV module via the USB and as it starts to communicate with the module the program says taulabsexe. has stopped working and that Windows will close the program shortly.
I'm baffled as to why it has chosen to stop working because the module work's fine and ive changed nothing on my computer. I also try to connect the module to another computer running Windows XP and the same happens.
A similar problem occurred a month ago after I installed Java. So I uninstalled Java but that didn't seem to fix the problem so I did a system restore and everything worked fine. Now a month later it happens again but tried another restore but it still refuses to work! Any idea's? Could it be any of Windows updates that may have changed something? Any help would be fanatastic!

tracernz, 2015-09-01:
Does it crash if you hit rescue in the uploader tab, and then plug it in? If that works, then try clicking boot to boot the board and see what happens.

Icemel, 2015-09-02:
Hi Michael, 

Thank's very much for replying to my question and so fast!

Yes when I click rescue and plug the device in the USB the program identifies it, says connection to bootloader successful. Then as soon as I hit "Boot" the same happens it says taulabsgcs.exe has stopped working and closes down!
Have you anymore suggestions or ideas that I could try? If you have when you get some spare time please let me know!  

Kindest regard's
Mel

Melvin Degraaf

tracernz, 2015-09-02:
I suspect perhaps one of the settings is stopping it from booting. If you rescue it, and look to the right side you will see the partition browser. You can right click the settings partition and download it to the PC (to back it up) and then right click -> erase that partition, now you will be booting with default settings.

Icemel, 2015-09-02:
Ok I saved that on file then erased partition settings, I then clicked boot but that closed down the program again.
I plugged in my quad and looked at the Brain's HUD on my FPV monitor which says CONFIG MANUAL ACTUATOR so that obviously worked but still something shuts the program down.
Thanks again for taking time to help me out:-)

Kindest regards,

Melvin Degraaf

mlyle, 2015-09-02:
@BrainFPV any ideas?

Icemel, 2015-09-03:
Just wondering should I erase "if possible" all other partitions Log, Metadata, Firmware? Or just erase Firmware and reinstall it?
Or maybe erase and wipe everthing off the Brain's board/memory and start over, re-install the Brain's Firmware and work up from that?
Please let me know if anyone else has had this problem. 
I've put a fair few of these Brain's on various customers Quad's and Multi's and this is the first major problem I just cant get my head around and it's on my best Quad!😪

mlyle, 2015-09-03:
@Icemel do you know / are you able to tell if it can talk to any other flight boards?  Or do you have just one? 

We haven't heard any similar bug reports.  Would be interesting to know if it is the GCS being broken in general, or if it just can't talk to this one specific board.

Icemel, 2015-09-03:
I'm stumped, I can flash the board with all the different Firmware, I can download and upload settings, log, all the partition's etc though im not sure what metadata and bootloader are.

I dont have access to another Brain cos most my build's have gone interstate! But i need to order more Brain's in the next month or so for other builds so I can differ them then if I have no luck!

Michael, thank's heap's for all your help, I very much appreciate your time and support! 
If I manage any progress in the meantime I'll be sure to post it! Thanks!!

BrainFPV, 2015-09-03:
@Icemel I'm not sure what could be causing the problem. I have many reports from people sucessfully using the GCS on Windows 10. Can you try deleting the `C:\Users\<user name>\AppData\Roaming\TauLabs\` directory? This makes sure the GCS uses the defaults. Also, does it work if you plug in the Brain before the GCS starts?

Icemel, 2015-09-04:
I've been using the Brain with Windows 10 for about 3 months now and have had no problem's with it so it cant be that, I have also plugged the Brain into my other computer running the old XP but the same happen's the program shuts down. 
If I plug the USB in before I start the program nothing happen's. Then the program starts but it doesn't connect or identify the device properly, as soon as I hit "connect" it shuts the program down! 
Also if i start the program (after) I plug the device into the USB, again it doesn't quite connect but as soon as I pull the USB plug out the program shuts down again.

Melvin Degraaf

Armadill0, 2015-09-23:
I also had exactly the same issues with my former CC3D using Taulabs GCS July release on Windows 10! It worked quite good some weeks and suddenly crashed some day while nothing on the system has been changed by myself.

First I thought that it could be an issue with the automated silent driver updates (which already caused issues with my Synaptics Touchpad driver before) by Windows 10. I tried to change the driver for the serial interface to the one which comes with Taulabs, which caused me to use some hacks to allow non signed drivers (https://www.youtube.com/watch?v=StkR3D2d5WI) for Windows 10. But it didn't help.

So I reinstalled Windows 10, which was necessary because of other reasons, too. :disappointed: