TauLabs/TauLabs #1891: Continued investigation of -ffast-math and other compiler flags needed

mlyle, 2015-09-02:
Also things like -fconserve-stack, -Os, etc..

kubark42, 2015-09-02:
@BrainFPV just made the excellent comment that although we should be unit testing for regressions due to compiler flags, our unit tests use a different compiler running on a different platform. 

Is there some kind of soft ARM core that we could run in emulation as part of the unit tests?

mlyle, 2015-09-02:
Yes, yes there is.

http://cgi.cs.indiana.edu/~geobrown/stm32/Main/Simulation

On the other hand, qemu is hard to drive and it's not the same as real hardware.  When you do a floating point cortex-M4 instruction, it runs a few instructions on the host processor and uses the host FPU.  So things like underflow behavior and precision can possibly be different.  (e.g. x86 is 80 bit internally, m4 is 32 bit).

It's also a bit difficult to set up a unit test environment that uses this.  But yes, this would be very good.