TauLabs/TauLabs #1897: Objects not getting updated over radio telemetry

mlyle, 2015-09-02:
From @pug398:

> I compared 20150819 f6593076eb (broken telem) against the master 2015817 b5c70e3599 (working) because I thought it would more pinpoint the problem but if peabody could have custom changes then maybe not. I will back up to next 20150812 6b2f4f4de5 as new starting point. If that doesn't work then I will go back further. Windows build of course :)

> Testing is straight forward. Apply firmware to fc with working radio telemetry and make some vpf changes and save. Disconnect and restart gcs which populates (vpf) with defaults. Connect via radio telemetry and on linkup uavo should transfer and update settings to match the ones on fc. Non working does not update the settings via radio telem. usb works.

> I was testing with sparky2 using both taulink and oplink mini with matching pipxteme. A test with quanton/xbee would eliminate a couple more variables and will try to tonight.

> Sucks to be at work when there is way more important things like testing :)

pug398, 2015-09-02:
If someone can test using sparky2/taulink/oplink and current next. Again make minor changes to vtolpathfollower settings and save via usb. Disconnect and restart GCS to get vpf default settings. Then connect via radio telemetry and see if it pulls in new changes. I am almost back up to current release with quanton/xbee and uavo are updating like they should. This may not be as global as I originally thought and could be more target specific. If it all works great for everyone then I would like to hear that too.

mlyle, 2015-09-02:
I am not able to test this second.  But one thing I'd note:  Recently I saw bad performance with bluetooth, much worse than I remembered.  I blamed it on the bluetooth radio being in a bad place and possible poor radio connection.  It did seem to work, but took 20-30 seconds to get the session fully connected.

pug398, 2015-09-02:
Back testing again now. 
Sparky2/TauLink and 20150901-78014174 will not transfer all uavo.
Quanton/xbee and 20150901-78014174 will transfer all uavo.

lolorc, 2015-09-02:
@pug398 how is it with quanton and a 115200 bps serial cable ? doesn't work at all here, gcs doesn't even connect to it. (while it works at 57600)

pug398, 2015-09-03:
You must set the GCS serial telemtry speed to 115k as it defaults to 57k. Tools>Options>Serial Telemetry. There may be more to it than forcing port speed.

pug398, 2015-09-03:
testing sparky2/taulink

next 20150812-6b2f4f4de5 transfers all uavo over air in ~20 seconds
next 20150819-f6593076 does not transfer all uavo

mlyle, 2015-09-03:
There's 9 commits in that range, including the 2 you've tested. 

So two likely things remain in that list:

HAL Refactor -- 4114f5cc (commit before this is 507dc3cbd)
GCS Compile with MSVC change -- 6d91a3d9 (commit before this is 810100c)  It looks like this has fallen off the Jenkins list.  The change right after it, 39fff898, is.

I am going to be able to test and bisect after my kids go to sleep in an hour or so.

pug398, 2015-09-03:
I would have put you on the exact one but the windows builder was broke :(

mlyle, 2015-09-03:
Test procedure:
* Checkout, make all_clean, make -j8 all
* Open GCS, rescue mode, connect flight board.  erase settings partition.  Flash matching firmware, boot.
* Confirm vtolpathfollowersettings missing
* Enable vtolpathfollower, unplug/replug USB
* Confirm vtolpathfollowersettings present
* Set loiterinitialmaxvel to 6.5.  Store to board.  Cycle USB and confirm still there.
* Disconnect board, Rescue, flash pipxtreme. boot.
* Run binding wizard (pipX coord).  64000/100/telemetry only
* Quit GCS.  Disconnect flight board (connected during binding wizard)
* Start GCS, connect pipxtreme.
* Power flight board
* Check vtolpathfollowersettings-- does it refresh?  Does loiterinitialmaxvel == 6.5?

Test one.  Confirm repro scenario.  a6554ba4   Sparky2 + original pipxtreme, matched revisions.  Confirmed broken.

Also confirmed all objects smaller than 54 bytes could be manually retrieved.  Objects 57 bytes or larger could not.

mlyle, 2015-09-03:
Test two.  Bisect at HAL refactor.  4114f5c Sparky 2 + original pipxtreme.  Confirmed broken.

Noticed that pipxtreme had larger UART buffers than all other targets and PIOS_HAL undid this.

Going to get less rigorous and just try a fix.

mlyle, 2015-09-03:
I believe this is fixed, hopefully @pug398 can confirm.  

@lolorc if you have issues with setting telemetry baud rate that are not fixed by changing the appropriate GCS setting, please open an issue so it can be looked at.

lolorc, 2015-09-03:
@pug398 / @mlyle  yes was indeed a pebcak problem, hopefully i went to bed straight away :)

PR #1901 (2015-09-04): pipxtreme: force usb rx/tx buf to 256
Unknown why this is necessary-- exceeds the 65 of other targets.

Presumably the radiocombridge module doesn't like buffering stuff itself.

Confirmed to work and resolve #1897 between sparky2 & pipxtreme.
closed 2015-09-04 by mlyle