TauLabs/TauLabs #1902: Purpose of discoveryf4 target?

mlyle, 2015-09-03:
I see we have just one target that's for something without sensors-- discoveryf4.

What's the purpose of this target?

Can it be abused to become an integration test target and have some tests put into it?

Relates to https://github.com/TauLabs/TauLabs/issues/1891#issuecomment-137176399

mlyle, 2015-09-03:
http://gnuarmeclipse.livius.net/wiki/How_to_use_QEMU

```
GNU ARM Eclipse QEMU v2.1.92 (qemu-system-gnuarmeclipse).
Supported machines are:
...
STM32F4-Discovery    ST Discovery kit for STM32F407/417 lines (Experimental)
...
```

kubark42, 2015-09-04:
It was added because it was the big brother of the DiscoveryF3 and we wanted to open the door up to innovation. Several people made shields (check out discussions on the old Phoenix Pilot Google Groups board) that would allow the DiscoveryF4 to turn into a FlyingF4. Ultimately, the FlyingF3 worked great and so a user base didn't really develop around DiscoveryF4. I don't believe there is any current usage of the DiscoveryF4.

If it were going to turn into an integration test target, we could rename it to the TestingF4. ;)

mlyle, 2015-09-04:
Yah, that's kind of what I'm inclined to do.  We have flyingf4 also for a predetermined set of sensors (@fujin may be building one).  discoveryf4 is kinda nice as a template for what a target looks like without (much) flight code, but maybe integration test can be cleanly added to it without making that too ugly.

doctorseus, 2015-09-15:
Aeroquad32 (aq32) is also (roughly) based on the STM32F4Discovery but is using a STM32F405 instead of the STM32F407 without all the fancy pancy addons. (Ethernet etc...)

mlyle, 2015-09-25:
Sadly,

```
Michaels-MBP:taulabs mlyle$ /Applications/GNU\ ARM\ Eclipse/QEMU/2.3.50-201508041609-dev/bin/qemu-system-gnuarmeclipse -board STM32F4-Discovery --image build/ef_discoveryf4/ef_discoveryf4.bin
qemu-system-gnuarmeclipse: Attempt to set CP10/11 in SCB->CPACR, but FP is not supported yet.
```

Looks like we won't be able to use the GNU ARM Eclipse QEMU distribution yet, at least for F3/F4.