TauLabs/TauLabs #1904: Radiocombridge's reliance on larger buffer sizes should be eliminated if possible

mlyle, 2015-09-03:
The radiocombridge module attempts to do nonblocking sends of telemetry data, which only works if the buffer can completely accept the data at once.  This requires any target that uses radiocombridge to have larger configured USB telemetry buffers than other targets, which use blocking sends in the normal telemetry module.

It's desirable to be able to use other targets as radiocombridges, and not to have this hidden dependency.

The most probable reason for the difference, as far as I can tell, is to atomically send (all or nothing) the data from the radio, without other UAVO's we're sending ourselves ending up in the middle.  PIOS_COM_SendBuffer does not acquire a lock, so if there are multiple senders data can end up interleaved.

mlyle, 2015-09-03:
Note the current synchronization mechanism is insufficient.  If pipxtreme's own telemetry has to block, there's a chance that an entire radio telemetry frame could end up in the middle of it.

mlyle, 2015-09-03:
Noticed re: #1901, #1897.

kubark42, 2015-09-04:
As a side note, if this problem proves intractable it would be good to implement a long-term solution with regards to https://github.com/TauLabs/TauLabs/pull/1901#issuecomment-137449469 and https://github.com/TauLabs/TauLabs/pull/1901#issuecomment-137489228.

mlyle, 2015-09-04:
Yup.  We need to make all the real target-specific differences nice and obvious.

I'm working to deduplicate code.. because right now, even with it in the c code, it's not obvious what's really different.