TauLabs/TauLabs #1905: debs produced by make package do not work on recent ubuntu

mlyle, 2015-09-03:
Are the linux package rules now superfluous?  We don't distribute it.

```
mlyle@stuffy:~$ /usr/bin/taulabs-gcs 
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: linuxfb, minimal, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)

mlyle@stuffy:~$ strace /usr/bin/taulabs-gcs
...
open("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/tls/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/tls/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/tls/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/tls", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/TauLabs/lib/taulabs/qtplugins/platforms/../../lib", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=102849, ...}) = 0
mmap(NULL, 102849, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f823d540000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/tls/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/x86_64-linux-gnu/tls/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/tls/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/x86_64-linux-gnu/tls", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/x86_64-linux-gnu/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/x86_64-linux-gnu", {st_mode=S_IFDIR|0755, st_size=12288, ...}) = 0
open("/usr/lib/x86_64-linux-gnu/tls/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/x86_64-linux-gnu/tls/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/tls/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/x86_64-linux-gnu/tls", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/x86_64-linux-gnu/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/x86_64-linux-gnu", {st_mode=S_IFDIR|0755, st_size=69632, ...}) = 0
open("/lib/tls/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/tls/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/lib/tls/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/tls", 0x7ffcdcdb87c0)        = -1 ENOENT (No such file or directory)
open("/lib/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib/x86_64", 0x7ffcdcdb87c0)     = -1 ENOENT (No such file or directory)
open("/lib/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
open("/usr/lib/tls/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/tls/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/tls/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/tls", 0x7ffcdcdb87c0)    = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/x86_64", 0x7ffcdcdb87c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/libQt5XcbQpa.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
...
```

mlyle, 2015-09-04:
Examining the package shows that many QT5 files are there, but not plugins like this:

```
mlyle@stuffy:~/TauLabs/build/test$ find . | grep so.5
./usr/local/TauLabs/lib/taulabs/libQt5Qml.so.5
./usr/local/TauLabs/lib/taulabs/libQt5XmlPatterns.so.5
./usr/local/TauLabs/lib/taulabs/libQt5DBus.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Svg.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Quick.so.5
./usr/local/TauLabs/lib/taulabs/libQt5OpenGL.so.5
./usr/local/TauLabs/lib/taulabs/libQt5SerialPort.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Gui.so.5
./usr/local/TauLabs/lib/taulabs/libQt53DRenderer.so.5
./usr/local/TauLabs/lib/taulabs/libicudata.so.54
./usr/local/TauLabs/lib/taulabs/libQt5XcbQpa.so.5
./usr/local/TauLabs/lib/taulabs/libQt5PrintSupport.so.5
./usr/local/TauLabs/lib/taulabs/libQt53DInput.so.5
./usr/local/TauLabs/lib/taulabs/libicui18n.so.54
./usr/local/TauLabs/lib/taulabs/libicuuc.so.54
./usr/local/TauLabs/lib/taulabs/libQt5Sql.so.5
./usr/local/TauLabs/lib/taulabs/libQt53DCore.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Core.so.5
./usr/local/TauLabs/lib/taulabs/libQt5QuickParticles.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Widgets.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Xml.so.5
./usr/local/TauLabs/lib/taulabs/libQt5Network.so.5
```

tracernz, 2015-09-04:
It would be nice to fix this up and use it for Pyreneus. I wonder if it adds a launcher shortcut on Ubuntu?