TauLabs/TauLabs #1911: Jenkins: All artifacts missing AQ32 and Naze firmware

tracernz, 2015-09-04:
I guess Jenkins must be using a custom command to gather up and zip the artifacts? It might be a good idea to use a standard make target so we can maintain it more easily (add a suitable one if required). In any case can we get these added please?

tracernz, 2015-09-04:
@lilvinz Can you please take a look at this? :)

mlyle, 2015-09-04:
Looks like the aq32/naze firmware blobs are still missing.  This is the last thing blocking a snapshot build of the next release.

lilvinz, 2015-09-12:
I have added aq32 and naze32 to the build scripts. A build of next is in progress.

lilvinz, 2015-09-12:
@tracernz You can open another issue for getting rid of the jenkins packaging scripts by adding a new or modifying an existing make target for that purpose.

closed 2015-09-12 by lilvinz

tracernz, 2015-09-12:
Thanks @lilvinz ! :-)