TauLabs/TauLabs #1913: Jenkins: Linux amd64 builds, GCS don't start up

LinuxJohannes, 2015-09-04:
Latest Master is working fine. But all latest NEXT on Jenkins don't work.

Output from the command line:
jh@linux-cjtf:~/TL/taulabs_next_20150904_013719_f30055fd4e_linux_amd64/gcs/bin> ./taulabsgcs
"fusion"
Starting the Udev client
*** addAction(): Unknown group:  "Bind Wizard"
UavTalkRelay SERVER listening on  QHostAddress("0.0.0.0") 2000
Loaded stylesheet: "/home/jh/TL/taulabs_next_20150904_013719_f30055fd4e_linux_amd64/gcs/share/taulabs/stylesheets" "default"
"UAVGadgetInstanceManager::createOptionsPages - failed to create options page for configuration TelemetrySchedulerGadget:default, configuration will be removed."
void PfdQmlGadgetWidget::setQmlFile(QString) "/home/jh/TL/taulabs_next_20150904_013719_f30055fd4e_linux_amd64/gcs/share/taulabs/pfd/default/Pfd.qml"
Cannot open file 'pfd.svg', because: No such file or directory
file:///home/jh/TL/taulabs_next_20150904_013719_f30055fd4e_linux_amd64/gcs/share/taulabs/pfd/default/SvgElementImage.qml:18: TypeError: Cannot read property 'height' of undefined
file:///home/jh/TL/taulabs_next_20150904_013719_f30055fd4e_linux_amd64/gcs/share/taulabs/pfd/default/SvgElementImage.qml:17: TypeError: Cannot read property 'width' of undefined
GLC_Context::GLC_Context
GLC_Context::chooseContext
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
enable to create context  0x2715900
ModelView: aircraft file not found.
Board not found
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Using Telemetry parser
Abgebrochen
jh@linux-cjtf:~/TL/taulabs_next_20150904_013719_f30055fd4e_linux_amd64/gcs/bin>

mlyle, 2015-09-04:
What graphics card do you have?  Does glxgears work with hardware acceleration?

mlyle, 2015-09-04:
Initially put into endomorphism release for triage.

LinuxJohannes, 2015-09-04:
It's a AMD Radeon HD 6550D with a AMD hardware acceleration driver, not the generic kernel driver.

LinuxJohannes, 2015-09-05:
Build this morning my own gcs from current next. It is working. But did a shortfix in textbubbleslider.h, QDesignerExportWidget not found.
//#include <QtUiPlugin/QDesignerExportWidget>
#include "/home/jh/Qt5.5.0/5.5/gcc_64/include/QtUiPlugin/qdesignerexportwidget.h"

LinuxJohannes, 2015-09-05:
And removed Sparky target, compile not possible : /opt/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld: /home/jh/workspace/TauLabs/TauLabs/build/fw_sparky/fw_sparky.elf section `.data' will not fit in region `flash'
/opt/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 240 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [/home/jh/workspace/TauLabs/TauLabs/build/fw_sparky/fw_sparky.elf] Error 1
make: *** [fw_sparky_tlfw] Error 2

kubark42, 2015-09-05:
@LinuxJohannes I notice that you're using an older gcc compiler. I can't say if this is related to the Sparky build failure or not, but you might try ```make arm_sdk_install``` and see if a clean build still results in the same problem. If it doesn't make a difference, then you can delete the local arm gcc directory from ```tools/gcc-arm-none-eabi-4_9-2015q1``` and it won't affect your system-wide ```/opt/gcc-arm-none-eabi-4_7-2013q3``` install.

LinuxJohannes, 2015-09-05:
OK, will do

LinuxJohannes, 2015-09-05:
@kubark42, you are right. solved with the the newer arm-compiler

mashu, 2015-09-05:
Same problem here on Debian 8.1
Linux thinkpod 4.1.0-2-amd64 #1 SMP Debian 4.1.6-1 (2015-08-23) x86_64 GNU/Linux

I did
```
make qt_sdk_install
make arm_sdk_install
make all
make package
dpkg -i dpkg -i build/taulabs_0.1.0-20150904-f30055fd_amd64.deb
```

latest commit is

```
~/TauLabs$ git log | head -5
commit f30055fd4e414bd4697e9640561b2b02185fa5d9
Merge: 8a8f581 e717a63
Author: Michael Lyle <mlyle@lyle.org>
Date:   Thu Sep 3 18:37:19 2015 -0700
```

When I start installed package I get
```
~/TauLabs$ taulabs-gcs 
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: linuxfb, minimal, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted
```

and when started locally

```
~/TauLabs$ build/ground/gcs/bin/taulabsgcs
"fusion"
Starting the Udev client
*** addAction(): Unknown group:  "Bind Wizard"
UavTalkRelay SERVER listening on  QHostAddress("0.0.0.0") 2000
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method
"Looking for default config files in: /home/mashu/TauLabs/build/ground/gcs/share/taulabs/default_configurations"
Load default config from resource  "/home/mashu/TauLabs/build/ground/gcs/share/taulabs/default_configurations/TauLabs_developer.xml"
Loaded stylesheet: "/home/mashu/TauLabs/build/ground/gcs/share/taulabs/stylesheets" "default"
"UAVGadgetInstanceManager::createOptionsPages - failed to create options page for configuration TelemetrySchedulerGadget:default, configuration will be removed."
void PfdQmlGadgetWidget::setQmlFile(QString) "/home/mashu/TauLabs/build/ground/gcs/share/taulabs/pfd/default/Pfd.qml"
Cannot open file 'pfd.svg', because: No such file or directory
file:///home/mashu/TauLabs/build/ground/gcs/share/taulabs/pfd/default/SvgElementImage.qml:18: TypeError: Cannot read property 'height' of undefined
file:///home/mashu/TauLabs/build/ground/gcs/share/taulabs/pfd/default/SvgElementImage.qml:17: TypeError: Cannot read property 'width' of undefined
GLC_Context::GLC_Context
GLC_Context::chooseContext
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
enable to create context  0x1cf19c0
ModelView: aircraft file not found.
Board not found
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Using Telemetry parser
Aborted
~/TauLabs$ 
```

PTDreamer, 2015-09-05:
What is the output of glxinfo?
What is the output of ldd?
Not a VM right?

mlyle, 2015-09-05:
Deb packages are broken; this is documented in issue #1905.  Unfortunately, the build infrastructure 'make package' is not used to produce actual release package builds, which we distribute as tar.xz.

The OpenGL problem on some linux systems id distinct.  I agree that ldd of the taulabsgcs binary and glxinfo would be helpful.

tracernz, 2015-09-05:
> #include "/home/jh/Qt5.5.0/5.5/gcc_64/include/QtUiPlugin/qdesignerexportwidget.h"
I'd guess that's not found because your Qt installation is neither in the TL tools directory nor on your system include path. This header changed location recently in the Qt tree. The older headers on your system include path probably worked for all of the others except this one.

mashu, 2015-09-06:
* The Qt is installed in TauLabs/tools/Qt5.5.0/ as recommended by console upon make qt_sdk_install (maybe it would be good to make it more automatic, so one does not need to rewrite path)

* The ldd of the binary

```
ldd TauLabs/build/ground/gcs/bin/taulabsgcs.bin 
	linux-vdso.so.1 (0x00007ffdd69bf000)
	libUtils.so.1 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libUtils.so.1 (0x00007fd775a17000)
	libExtensionSystem.so.1 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libExtensionSystem.so.1 (0x00007fd7757ec000)
	libAggregation.so.1 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libAggregation.so.1 (0x00007fd7755e7000)
	libQt5Widgets.so.5 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Widgets.so.5 (0x00007fd774d69000)
	libQt5Gui.so.5 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Gui.so.5 (0x00007fd774555000)
	libQt5Network.so.5 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Network.so.5 (0x00007fd7741fc000)
	libQt5Xml.so.5 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Xml.so.5 (0x00007fd773fc0000)
	libQt5Core.so.5 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Core.so.5 (0x00007fd77387a000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007fd7735c1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd7733a4000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd773028000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd772d27000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd772b11000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd772767000)
	libQt5Svg.so.5 => /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Svg.so.5 (0x00007fd772513000)
	libQt5OpenGL.so.5 => /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5OpenGL.so.5 (0x00007fd7722bb000)
	libQt5Quick.so.5 => /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Quick.so.5 (0x00007fd771ced000)
	libQt5Qml.so.5 => /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Qml.so.5 (0x00007fd7716bf000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd7714ba000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fd771268000)
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007fd771065000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd770e5d000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd770b4e000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fd77093b000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd7705f8000)
	libicui18n.so.54 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libicui18n.so.54 (0x00007fd770189000)
	libicuuc.so.54 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libicuuc.so.54 (0x00007fd76fddb000)
	libicudata.so.54 => /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libicudata.so.54 (0x00007fd76e3b1000)
	/lib64/ld-linux-x86-64.so.2 (0x0000559425318000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fd76e187000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007fd76df59000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007fd76dd56000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fd76db4f000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fd76d94d000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007fd76d734000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007fd76d52e000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007fd76d32b000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007fd76d128000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007fd76cf20000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fd76ccfe000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007fd76cafb000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fd76c8f4000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007fd76c6e6000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fd76c4dc000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fd76c26e000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fd76c069000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fd76be63000)
```

* Part of glxinfo

```
glxinfo 
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
[... 16 lines cut ...]
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
[... cut ...]
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) 965GM 
OpenGL version string: 2.1 Mesa 10.6.5
OpenGL shading language version string: 1.20
...etc..
```

* Not a VM, I don't have other OS

I know Debian building works in OpenPilot GCS, shouldn't the projects collaborate or borrow from each other ? Or at least maybe it's worth looking at how they do it ? Also they don't have issue with OpenGL on this computer. Maybe that could help somehow to improve Taulabs GCS for Linux.

mlyle, 2015-09-06:
Backtrace:

```
Program received signal SIGABRT, Aborted.
0x00007fe50ce7f107 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007fe50ce7f107 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007fe50ce804e8 in __GI_abort () at abort.c:89
#2  0x00007fe50dfcc0be in QMessageLogger::fatal(char const*, ...) const ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Core.so.5
#3  0x00007fe50c535b64 in QSGRenderLoop::handleContextCreationFailure(QQuickWindow*, bool) ()
   from /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Quick.so.5
#4  0x00007fe50c536c93 in ?? () from /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Quick.so.5
#5  0x00007fe50c5378de in ?? () from /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Quick.so.5
#6  0x00007fe50ed0b6a5 in QWindow::event(QEvent*) () from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Gui.so.5
#7  0x00007fe50c570dbb in QQuickWindow::event(QEvent*) () from /home/mashu/TauLabs/tools/Qt5.5.0/5.5/gcc_64/lib/libQt5Quick.so.5
#8  0x00007fe50f57c4dc in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Widgets.so.5
#9  0x00007fe50f581640 in QApplication::notify(QObject*, QEvent*) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Widgets.so.5
#10 0x00007fe50e1cb6a3 in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Core.so.5
#11 0x00007fe50ed03e70 in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Gui.so.5
#12 0x00007fe50ed04afd in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Gui.so.5
#13 0x00007fe50ece9c4f in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Gui.so.5
#14 0x00007fe505ded170 in ?? () from /home/mashu/TauLabs/build/ground/gcs/lib/taulabs/libQt5XcbQpa.so.5
#15 0x00007fe50b27dc3d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007fe50b27df20 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007fe50b27dfcc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007fe50e220ae7 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Core.so.5
#19 0x00007fe50e1c9252 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Core.so.5
#20 0x00007fe50e1d0d3d in QCoreApplication::exec() () from /home/mashu/TauLabs/build/ground/gcs/bin/../lib/taulabs/libQt5Core.so.5
#21 0x000000000040aafe in main ()
```

mlyle, 2015-09-06:
So, time to sleep, but some observations:

1. This is crashing in the qtquick stuff, so it's not as easy as disabling modelview to get this to work.
2. This is failing because the appropriate GL contexts fail to build.  Selection between egl and glx used to be compile time but Qt5.5 has adapted this to a dynamic routine.
3. @tracernz says it works with his own builds but fails with jenkins builds.  This suggests one route for investigation.  (I should try jenkins builds on my Linux machine; I have not had a chance yet).
4. I was concerned there could be problems getting the wrong QT libraries loaded, but it looks like from an investigation of `info sharedlibrary` on tracernz's system on the Jenkins build looks like that's not the case.

LinuxJohannes, 2015-09-06:
* System Opensuse 13.1
* The Qt is installed in  ~/Qt5.5.0/ as recommended by console upon make qt_sdk_install 
* The ldd of my own build binary

        ldd TauLabs/build/ground/gcs/bin/taulabsgcs.bin
        linux-vdso.so.1 (0x00007ffef51b9000)
        libUtils.so.1 => /home/jh/workspace/TauLabs/TauLabs/build/ground/gcs/bin/../lib/taulabs/libUtils.so.1 (0x00007f69fced4000)
        libExtensionSystem.so.1 => /home/jh/workspace/TauLabs/TauLabs/build/ground/gcs/bin/../lib/taulabs/libExtensionSystem.so.1 (0x00007f69fcc87000)
        libAggregation.so.1 => /home/jh/workspace/TauLabs/TauLabs/build/ground/gcs/bin/../lib/taulabs/libAggregation.so.1 (0x00007f69fca7d000)
        libQt5Widgets.so.5 => /usr/lib64/libQt5Widgets.so.5 (0x00007f69fc206000)
        libQt5Gui.so.5 => /usr/lib64/libQt5Gui.so.5 (0x00007f69fbaa3000)
        libQt5Test.so.5 => /usr/lib64/libQt5Test.so.5 (0x00007f69fb875000)
        libQt5Network.so.5 => /usr/lib64/libQt5Network.so.5 (0x00007f69fb524000)
        libQt5Xml.so.5 => /usr/lib64/libQt5Xml.so.5 (0x00007f69fb2e6000)
        libQt5Core.so.5 => /usr/lib64/libQt5Core.so.5 (0x00007f69fac21000)
        libGL.so.1 => /usr/X11R6/lib64/libGL.so.1 (0x00007f69faa24000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f69fa806000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007f69fa4fe000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f69fa1fb000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f69f9fe4000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f69f9c36000)
        libQt5Svg.so.5 => /usr/lib64/libQt5Svg.so.5 (0x00007f69f99e0000)
        libQt5OpenGL.so.5 => /usr/lib64/libQt5OpenGL.so.5 (0x00007f69f9786000)
        libQt5Quick.so.5 => /usr/lib64/libQt5Quick.so.5 (0x00007f69f91a9000)
        libQt5Qml.so.5 => /usr/lib64/libQt5Qml.so.5 (0x00007f69f8b8e000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f69f898a000)
        libgobject-2.0.so.0 => /usr/lib64/libgobject-2.0.so.0 (0x00007f69f8739000)
        libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007f69f8436000)
        libX11.so.6 => /usr/lib64/libX11.so.6 (0x00007f69f80f8000)
        libpng16.so.16 => /usr/lib64/libpng16.so.16 (0x00007f69f7ebb000)
        libz.so.1 => /lib64/libz.so.1 (0x00007f69f7ca5000)
        libssl.so.1.0.0 => /lib64/libssl.so.1.0.0 (0x00007f69f7a3d000)
        libcrypto.so.1.0.0 => /lib64/libcrypto.so.1.0.0 (0x00007f69f7651000)
        libicui18n.so.51.2 => /usr/lib64/libicui18n.so.51.2 (0x00007f69f7240000)
        libicuuc.so.51.2 => /usr/lib64/libicuuc.so.51.2 (0x00007f69f6ec3000)
        libpcre16.so.0 => /usr/lib64/libpcre16.so.0 (0x00007f69f6c60000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f69f6a58000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f69fd1c8000)
        libXext.so.6 => /usr/lib64/libXext.so.6 (0x00007f69f6846000)
        libffi.so.4 => /usr/lib64/libffi.so.4 (0x00007f69f663e000)
        libpcre.so.1 => /usr/lib64/libpcre.so.1 (0x00007f69f63d1000)
        libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00007f69f61b2000)
        libicudata.so.51.2 => /usr/lib64/libicudata.so.51.2 (0x00007f69f5fb1000)
        libXau.so.6 => /usr/lib64/libXau.so.6 (0x00007f69f5dad000)

* ldd of the Jenkins binary
```
~/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin> ldd ./taulabsgcs.bin
        linux-vdso.so.1 (0x00007fff87f35000)
        libUtils.so.1 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libUtils.so.1 (0x00007f496dd96000)
        libExtensionSystem.so.1 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libExtensionSystem.so.1 (0x00007f496db6b000)
        libQt5Widgets.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Widgets.so.5 (0x00007f496d2ed000)
        libQt5Gui.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Gui.so.5 (0x00007f496cada000)
        libQt5Network.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Network.so.5 (0x00007f496c781000)
        libQt5Core.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Core.so.5 (0x00007f496c03c000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f496be1e000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007f496bb16000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f496b8ff000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f496b551000)
        libQt5Svg.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Svg.so.5 (0x00007f496b2fd000)
        libQt5Quick.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Quick.so.5 (0x00007f496ad2f000)
        libQt5Xml.so.5 => /home/jh/TL/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/../lib/taulabs/libQt5Xml.so.5 (0x00007f496aaf3000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f496a7f0000)
        libgobject-2.0.so.0 => /usr/lib64/libgobject-2.0.so.0 (0x00007f496a59f000)
        libgthread-2.0.so.0 => /usr/lib64/libgthread-2.0.so.0 (0x00007f496a39d000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f496a195000)
        libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007f4969e92000)                                                                                                                                                            [... cut ...]
OpenGL core profile shading language version string: 4.40                                                                                                                                                                                  
OpenGL core profile context flags: (none)                                                                                                                                                                                                  
OpenGL core profile profile mask: core profile     
...etc..                                                                                                                                                ```

tracernz, 2015-09-06:
Loaded libraries on my own working build: https://gist.github.com/tracernz/4d5e4d2267fc8eac2265
Broken Jenkins build: https://gist.github.com/tracernz/75ed4f5203e331aba846

tracernz, 2015-09-06:
> The Qt is installed in ~/Qt5.5.0/ as recommended by console upon make qt_sdk_install

This is a side issue but anyhow..
The QT SDK install dir is always:
```
ROOT_DIR is root of Tau Labs repo
TOOLS_DIR := $(ROOT_DIR)/tools
QT_SDK_DIR := $(TOOLS_DIR)/Qt5.5.0
```
The header your compiler is trying to find was shifted in Qt 5.5.0 qtproject/qttools@7d4149e61fbb299e95968da22daa0818e30802b9, and I switched us to the new header in 846c45a62b7148c4e99078a10c3aafec34dd5c71 hence why I think your compiler is using the system Qt headers(likely < 5.5, hence can't qdesignerexportwidget.h at the new location) rather than your Qt 5.5 headers.

mashu, 2015-09-06:
Ok, seems like I solved the problem. I had some missing dependencies to build firmware, particularly libc6-dev:i386 was missng and gcc:i386 since Debian uses multiarch.
Then instead of as suggested by [https://github.com/TauLabs/TauLabs/wiki/Development-Linux-setup#building-gcs](https://github.com/TauLabs/TauLabs/wiki/Development-Linux-setup#building-gcs)

```
make gcs
```
 I had to do 
```
make all
build/ground/gcs/bin/taulabsgcs
```
This is either coincidence or solved the problem. Either case, if GCS is indeed aborting without reporting error about missing firmware, it would be good to fix that and update documentation to reflect that firmware is needed and has to be build with **make all** prior to running gcs.
Also it would be good to fail package build in case of missing firmware.

mlyle, 2015-09-06:
GCS doesn't depend on firmware to start up-- I start without firmware all the time.  The abort() backtrace you sent me was a crash from a failure in QT to initialize an OpenGL context.

mlyle, 2015-09-07:
Various bad options to go forward and make a snapshot build

1. Use Jenkins artifacts
  * Impact: Some (perhaps all) Linux systems will be broken
  * Impact: Two supported targets will not be packaged in this snapshot on Linux & Windows
2. Manually produce build
  * Impact: Linux may be broken still (this seems most likely) but needs verification
  * Impact: Build will not be reproducible / made on my machines
3. Manually produce build and figure out how to alter to package missing QT components
  * Impact: Last minute hackery
  * Impact: Build will not be reproducible / made on my machine

Right now looks like choosing between options 2 & 3 is best.

tracernz, 2015-09-07:
@mlyle asked me to run with strace to see what might be going on. I found that it stats this directory and then does nothing further with it when it isn't found:
```
19100 stat("/home/mike/Desktop/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/lib/taulabs/qtplugins/xcbglintegrations/.", 0x7ffc7a5bdbe0) = -1 ENOENT (No such file or directory)
19100 stat("/home/mike/Desktop/taulabs_next_20150906_032821_8515a25c46_linux_amd64/gcs/bin/xcbglintegrations/.", 0x7ffc7a5bdbe0) = -1 ENOENT (No such file or directory)
```
Copying the xcbglintegrations directory from Qt5.5.0/5.5/gcc_64/plugins to <tl>/gcs/lib/taulabs seems to fix the Jenkins build.

mlyle, 2015-09-07:
```
Michaels-MBP:taulabs_next_20150907_144637_e0d3d51304_linux_amd64 mlyle$ find . | grep -i glinteg
./gcs/lib/taulabs/qtplugins/xcbglintegrations
./gcs/lib/taulabs/qtplugins/xcbglintegrations/libqxcb-glx-integration.so
```

Can someone please try the Jenkins e0d3d51 build to see if it's fixed?

tracernz, 2015-09-07:
Can confirm both the AMD64 and i686 builds work.
@LinuxJohannes ?

LinuxJohannes, 2015-09-08:
Yes, yes,
taulabs_next_20150907_144637_e0d3d51304_linux_amd64
works with out problems on my system. Thanks

jpahka, 2015-09-08:
I can also confirm, had a problem with jenkins builds on my 64bit arch linux earlier, suspected it to be a problem with my computer, but this fixes the gcs crash @ start for me as well.

tracernz, 2015-09-08:
Thanks for the feedback, appreciated. :-)

jpahka, 2015-09-08:
Thanks for the work, much appreciated ;)

mlyle, 2015-09-08:
Great!  If anyone has any remaining issues with OpenGL on Linux, please open a new issue.  Thanks everyone.  Fixed by #1920.

closed 2015-09-08 by mlyle