TauLabs/TauLabs #1915: Ariframe trimming (fixed wings)

robustini, 2015-09-04:
In my opinion this function it doesn't work in the latest official code, please check.
When replug the main power all the surface are in the position before the trimming function.

kubark42, 2015-09-04:
@robustini, sorry to hear you're having troubles. Could you describe your process in a little more detail so we can help you better? And in specific, are you using the "Set Level Angle" button or the "Trim servos" button?

robustini, 2015-09-04:
I use "Trim Servos" button in "manual" mode, Atom board.
I follow the step in the help popup.

kubark42, 2015-09-04:
> Everything is ok until i disconnect and reconnect the LiPo.

Aha! The problem is that the values aren't being saved. After following all the directions, go to the "Output" tab and click the Save button there. 

P.S. I realize this isn't intuitive, and we can probably come up with a better way of doing it. Thanks for bringing this to our attention. I'll leave this issue open so that we remember we to look at improving the UI.

robustini, 2015-09-04:
I'm sorry @kubark42, i've click "Apply and " "Save" button at the end, not written in the previous post.

mlyle, 2015-09-04:
@robustini -- did you go to the output tab before saving?

I understand it's a bug that you'd have to do that.

kubark42, 2015-09-04:
@robustini You need to go to the output tab and hit "Save" there.

robustini, 2015-09-04:
Ah ok, thanks!
Not too intuitive, is not possible save this trim value in the "Veichle" tab in a future gui? ;)

robustini, 2015-09-04:
Or move this function in the "Output" tab...

kubark42, 2015-09-04:
@robustini It's an airframe specific configuration, so from a UI perspective it's part of the Airframe tab. Our workflow, however, encourages you to save page by page so I understand your confusion, and is why I labeled it a bug. It really is broken from a user point of view.

I think the best solution is to have the Airframe tab save it directly, but I can see arguments for moving it to Outputs.

robustini, 2015-09-04:
@kubark42: thumbs up!