TauLabs/TauLabs #1919: Bad Mag drift than 1 month ago

ducminhkt, 2015-09-06:
I tested 2 version of TL to compare PH result: f30055f (04/09/15) and version c708667 (06/08/15). Both on the same hardware, same time, same position, same number of satellite. Newer version get bad result in PH mode than the older. With newer version, my quad move in circle (radius ~ 2m, and continue increase velocity), when the older, quad hold pretty well in radius about 1m, no move in circle.
Then I check the mag direction, with newer version, mag has more drift (~ 10 degrees when hover) than older version (~ 1-2 degrees when hover), same hardware config.

kubark42, 2015-09-07:
@ducminhkt That's not good. A lot of work has been going on lately, so let's see if we can get to the bottom of this quickly.

1. What board was this with?
2. Are the results easily reproducible? Did you get a chance to try toggling several times between firmwares and always see the same results?
3. Which state estimation filter where you using?
4. The complementary filter's yaw initialization code will return incorrect results when the board is inclined. Did you power on from a non-inclined position?

As a quick test remedy, I would suggest arming and disarming about 10 times before taking off. This helps the yaw converge to the magnetic compass heading.

It will also help us get to the bottom of things more quickly if you can test a few intermediate commits and see where the breakage first occurs and limit where we have to look. And it would be double-plus good if you could bisect it down until you have the exact pull request when the breakage occurs.

P.S. I could figure it out from the commits, but in the future you might try writing dates as YYYY-MM-DD. This is the only way to be sure if 04/09/15 means September 4th or April 9th. Plus, YYYY-MM-DD is the most useful date format since sorting numerically returns the desired results, whereas neither DD-MM-YYYY nor MM-DD-YYYY give chronological results when sorting by number.

mlyle, 2015-09-07:
@ducminhkt Also, did you reset configuration inbetween?

While there's been no change to the mag handling code, the default tuning for the vtolpathfollower velocity loop has changed.  The change is pretty conservative and has been verified on multiple platforms-- from a KP=3.0/KI=0 to KP=2.6/KI=0.4.

ducminhkt, 2015-09-08:
Thank you for answer.
@kubark42 1. "What board was this with?": I use FlyingF4. "Did you get a chance to try toggling several times between firmwares and always see the same results?": No, but I sure I did every step carefully to change firmware. 3: Complementary and INS, 4: I don't think it make problem because every config same as the older version. "It will also help us get to the bottom of things more quickly if you can test a few intermediate commits and see where the breakage first occurs and limit where we have to look: OK, I will. I use dd/MM/yyyy date format, so 04/09/15 means September 4th, sorry about that. I will use YYYY-MM-DD in the future.

@mlyle I alway do erase all setting from menu "Tools -> Erase all settings from board" each time I flashed a firmware version.

ducminhkt, 2015-09-08:
I just erase chip and upload new firmware build on 2015-09-07, then config and test. Now I don't have a good place for PH testing, so I only test mag drift by hold quad on ground and apply haft throttle. Seem mag drift not much. I attach some picture below. Please tell me how much degrees drift can cause PH bad result ?
Mag: disarm
[image]
mag: 50% throttle
[image]

Scope
[image]

kubark42, 2015-09-08:
@ducminhkt, it does indeed seem like the magnetometer isn't drifting very much. In my experience the magnetometer doens't feed into the filter very strongly with the default settings, so my initial thought is that that some of the instant change in yaw might be due to a slight physical airframe shift as you run the motors up and down.

In general, my rule of thumb is to keep misalignments less than ~5 degrees in order to avoid toilet-bowling. The graph you show is just on the cusp of that, so I don't yet know what to think. How did you determine that previously the yaw error was 1-2 degrees (excellent!) vs 10 degrees (yuck) now? Was it the same test as above? If so, could you run the test with 6d91a3d9f1c17dbab5ddb445937217aea7d2f223? There are no firmware changes from c708667, so it should perform identically.

@mlyle even if there haven't been any mag changes, there have been some gyro changes lately, right? Not saying their responsible, but it would be good to narrow down

peabody124, 2015-09-08:
I haven't had a chance to check if it is in the window, but INS13->INS14 introduced substantial changes in mag handling. They should generally make it less sensitive to magnetic disturbances in the environment, but I cannot rule out some cases that not being true.

mlyle, 2015-09-08:
> @mlyle even if there haven't been any mag changes, there have been some gyro changes lately, right? Not saying their responsible, but it would be good to narrow down

That's true, the gyro low pass filter was changed.  But flyingf4 is one of the 666Hz default-rate platforms which should see basically no change.

> I haven't had a chance to check if it is in the window, but INS13->INS14 introduced substantial changes in mag handling. 

Yah, this is way after that-- this release cycle.  The main relevant things I can see in the revision log are

* zippyloiter
* default vtol tuning from 3.0/0.0 to 2.6/0.4 PI gains

If it's just that there was a bad, non-repeated position hold flight...I think it's easy see substantial variation flight to flight based on GPS coverage, wind conditions, starting direction (sensitive to different parts of mag cal).

ducminhkt, 2015-09-09:
@kubark42 Yesterday I upload and test again with two firmware. Seem no change with mag drift. So I think maybe I did some wrong step previously. But I must do a test with PH in this weekend to confirm that.
@mlyle  "But flyingf4 is one of the 666Hz default-rate platforms which should see basically no change" I changed gyro rate to 2000, is it make difference ?

kubark42, 2015-09-09:
> During calibrate process, user hold platform on ground, with props, arm and apply half throttle (hover throttle). After some seconds, user disarm and we save the value changed of mag direction. Then we can use this value to adjust mag direction in navigation feature (PH, RTH, ...). 

The magnetic field distortion due to current will shift the center of the magnetic field, so it's not enough to provide an offset. The center needs to be recalculated, and that done as a function of current.

@mluessi did a few experiments with calibrating out the offset due to current, but the results weren't conclusive. They didn't seem to restore the good behavior of the mag before motor current was present.

mlyle, 2015-09-22:
@ducminhkt any more details on this bug?