TauLabs/TauLabs #1923: Problem with axis reverse code in MPU6K driver

AlessioMorale, 2015-09-09:


kubark42, 2015-09-09:
Oooh, that's a nasty one! We should check that this is restricted to only the mpu6000.

I see how you resolved it for https://github.com/librepilot/LibrePilot/blob/next/flight/pios/common/pios_mpu6000.c#L580. Would you still recommend this approach, or in the meantime has a better solution presented itself?

mlyle, 2015-09-09:
Yes, this is high severity.

mlyle, 2015-09-09:
I don't think we can ship with something like this known.

mlyle, 2015-09-09:
(I'm in favor of the -1 hack, but it's unfortunate we even treat these as signed quantities given that we need to find/apply an offset later anyways.)

AlessioMorale, 2015-09-09:
This is still fine. being a "digital" sensor 1 lsb does not make a lot of difference

Trex4Git, 2015-09-09:
In "axis = - (int16_t)original value;" in Taulabs "axis" is a float variable; I am not sure when  the conversion to float does happen. 
Maybe a more explicit "axis = - 1.0f *  (int16_t)original value;" might work, as I think the multiplication of a float and an int should be a float ?

kubark42, 2015-09-09:
> Maybe a more explicit "axis = - 1.0f * (int16_t)original value;" might work, as I think the multiplication of a float and an int should be a float ?

Yes, that is a very effective guard against a potential problem.

mlyle, 2015-09-09:
I will try to read the code and look at the types of all this stuff later tonight and open a PR.

PR #1925 (2015-09-12): pios_mpu*: Correct orientation math / prevent overflow
Signed wraparound has undefined behavior in C, and here it was worse,
where the obvious thing flipped sign for maximum values in one
direction.  Used @trex4git's suggestion of a floating point multiply to
ensure sign promotion.  Hat tip to @AlessioMorale for pointing out this
severe issue.

Fixes #1923.
commit 0f0b00a4 (2015-09-12): pios_mpu*: Correct orientation math / prevent overflow

Signed wraparound has undefined behavior in C, and here it was worse,
where the obvious thing flipped sign for maximum values in one
direction.  Used @trex4git's suggestion of a floating point multiply to
ensure sign promotion.  Hat tip to @AlessioMorale for pointing out this
severe issue.

Fixes #1923.
closed 2015-09-12 by kubark42