TauLabs/TauLabs #1926: UAVO: New HwShared parent causes field misalignment

kubark42, 2015-09-10:
Not quite sure where this is happening (ground? firmware? both?), so pardon me for an incomplete bug report.

Symptom:

I have a hardware UAVO which has a UART3 field with only five options. The fourth option is I2C. 

```
<field name="Uart3" units="function" type="enum" elements="1" parent="HwShared.PortTypes"  defaultvalue="Disabled">
	<options>
		<option>Disabled</option>
		<option>Telemetry</option>
		<option>GPS</option>
		<option>I2C</option>
		<option>DSM</option>
	</options>
</field>
```
When selecting I2C via the UAVO browser and saving, upon rebooting the UAVO browser populates the name "DSM". Inside HwShared.xml, DSM is the fourth option. When moving I2C to the third place in the list

```
<field name="Uart3" units="function" type="enum" elements="1" parent="HwShared.PortTypes"  defaultvalue="Disabled">
	<options>
		<option>Disabled</option>
		<option>Telemetry</option>
		<option>I2C</option>
		<option>GPS</option>
		<option>DSM</option>
	</options>
</field>
```

And performing the same save/reboot operation, the UAVO browser populates the name with GPS (the third option in HwShared.xml).

I don't know if this is only a UAVO browser problem, or if it is a bonafide firmware misconfiguration. I'll look into it later when I get a chance, but for the moment I opted to remove all options from the UART, thus using the default from HwShard and ensuring compatibility. Note, this is not a good workaround, as many of the options aren't supported by my board.

Only tested on in-house hardware, don't know if this is repeatable on officially supported targets. I suspect it is, though.

mlyle, 2015-09-10:
Argh.  Thanks for the report.  I originally suspected some of the combobox indexing code would have to change in the uavo browser to support parent fields, but I did some test and it seemed OK in practice.  I will come up with the minimal fix ASAP.

mlyle, 2015-09-11:
I am having trouble reproducing this.  Right now I'm playing with sparky2 which has a pretty limited set of options on RcvrPort.  Both the uavobject browser and hardware pane seem to do the right thing.

kubark42, 2015-09-11:
Hmm... Okay, I'll dive more into it when I get the chance. It might mean that I implemented the HAL changes incorrectly, or that there's something else like #1921 which is most easily seen on a new target.

The way I see it, if it works fine on Revolution and Sparky2, then that's 100% coverage for the HAL changes that are in this release, and thus we can remove the blocker label.

mlyle, 2015-09-11:
I will play more, but it doesn't need to stop a snapshot from going out.

mlyle, 2015-09-14:
I also appreciate any other test people can put on hwsparky2 or hwrevomini in the uavobject browser.

mlyle, 2015-09-17:
I'm planning on removing the blocker label and tagging this out of the release Friday if I've not heard anything new about this.

peabody124, 2015-09-17:
@mlyle want to remove the endomorphism tag as well?

kubark42, 2015-09-17:
I think you should remove the blocker label. I suspect that the outcome of this bug will be a more robust check for hardware misconfiguration as developers create new targets.

mlyle, 2015-09-19:
Removing blocker label and tagging out of release as promised.