TauLabs/TauLabs #1941: Fix PIOS_DELAY_WaitmS jitter

kubark42, 2015-09-14:
In https://github.com/TauLabs/TauLabs/blob/next/flight/PiOS/Common/pios_delay.c#L133, I noticed that the function is guaranteed to cause a tiny amount of jitter, since each ms of delay takes 1000us plus the time required to evaluate the while and the delay. 

It's tough to say how big a deal this is, since most of the time in the code we're waiting until the clock reaches a certain time, and when we are doing something like ```WaitmS()``` it's usually not important that the value be exact.

tracernz, 2015-09-14:
```c
PIOS_DELAY_WaituS(mS * 1000);
```
Should be okay as long as we don't wait more than 4294967.295 ms?

PS. Why uppercase s for seconds? S = Siemens, not seconds.

kubark42, 2015-09-14:
@tracernz The concern is that the 1ms loop takes an indeterminate amount of time longer than 1ms. It's probably not much-- if I had to guess I'd say it's on the order of just a few microseconds-- but it's still a source of jitter in a hard real-time system. I like to squash those when possible. :)

tracernz, 2015-09-14:
Yeah I got that, I was suggesting replacing the code in the function with `PIOS_DELAY_WaituS(mS * 1000)`, eliminating that while loop. ;)

peabody124, 2015-09-14:
Be very careful. The us delay method is much more precise, but it does that at the cost of spinning in a while loop and blocking other threads. It should be used with great caution!  A blind substitute like that through the code would break it.

@kubrak42 ultimately consider the RTOS real time on the 1-2 ms range. Anything more than that will require a great deal of attention, e.g. using specific timers. I'm not sure there's is a solution to this. You might be able to run the chibios tick counter at 10khz, but that would ultimately cost a lot...

tracernz, 2015-09-14:
The RTOS delay functions which put the thread to sleep are seperate to this AFAIK? I don't see how the while loop in PIOS_DELAY_WaitmS is any different to the one in PIOS_DELAY_WaituS from the RTOS point of view, both will block the CPU while the calling task is running. The RTOS will still context switch to another task at the end of the time-slice (1 ms with our current 1000 Hz tick-rate) as before, and you will still get jitter associated with that. Having thought about that perhaps there isn't much to gain from this really.

mlyle, 2015-09-14:
Yah, the very concept of a busywait dimensioned in milliseconds is completely broken, except for before the RTOS has started during device initialization.  And it's odd because this is a quasi-busywait.

The outer loop only gets dodgy when combined with task switching.  If 10 milliseconds goes by while other tasks run, the inner loop will notice this.  The outer loop will not give us credit for the extra 9 milliseconds.

tracernz, 2015-09-15:
Even the us delay doesn't disable the scheduler so if it tries to wait past the end of the time slice it could be several ms worst case. These two functions are really PIOS_DELAY_AtLeast(m|u)s() once the scheduler starts.

mlyle, 2015-09-15:
Yup.. And then the ms version can incur that penalty many times.

kubark42, 2015-09-15:
> @kubrak42 ultimately consider the RTOS real time on the 1-2 ms range. Anything more than that will require a great deal of attention, e.g. using specific timers. I'm not sure there's is a solution [...]

I get where you're coming from. I don't think we do anything right now that matters much for this, although timing upgrades always have their place. This jumped out at me because we had just solved a jitter problem that was exactly due to this kind of code structure. 

I do agree with @tracernz, though, that it seems that ```WaituS()``` and ```WaitmS()``` should have the same effect on the RTOS. Maybe that used to be different and now has changed with the ChibiOS-related abstractions? If it really is this low-hanging, then it might be worth a stab at changing.

mlyle, 2015-09-15:
Right now, I think:

* WaitmS is pretty good-- probably about 1 part per thousand-- when not task switching.  When not switching, the remainders in waitus for 1000 shouldn't be expected to be long, and to re-enter the wait the time is probably on the order of a couple hundred cycles and pretty deterministic.
* When we're task switching / in the RTOS it doesn't seem to ever make sense to quasi-busywait on a timescale of milliseconds.  We should audit and get rid of any code that does this.
* I agree the us/ms variants are equivalent from the point of view of the rtos, except the us variant counts time when other tasks are running, and the ms variant (mostly) doesn't.
* We might as well trivially improve the ms version.  IMO the way to do this is to just change the timescale of the while loop.  while (the time to wait > the time that can be safely waited in us version) { wait the time that can be waited in us version; }   then wait the remainder

peabody124, 2015-09-15:
Ignore my comment above. I was confusing the delay ms comment with the thread sleep 1ms command (which uses the RTOS). Teach me to reply from phone

kubark42, 2015-09-15:
Proposed change:

```
/**
 * Waits for a specific number of ms
 *
 * Example:<BR>
 * \code
 *   // Wait for 500 ms
 *   PIOS_DELAY_Wait_mS(500);
 * \endcode
 * \param[in] ms delay (0..4,294,967,296 milliseconds)
 * \return 0
 */
int32_t PIOS_DELAY_WaitmS(uint32_t ms)
{
// Define the wait window as 65 so that the call to PIOS_DELAY_WaituS()
// will fit into a uint16. This seems to be a happy compromise between
// potential future changes to the arguments for PIOS_DELAY_WaituS() and
// the desire not to call PIOS_DELAY_WaituS() more often than necessary.
#define MS_WAIT_WINDOW 65

	// wait() for the quotient
	while (ms > MS_WAIT_WINDOW) {
		PIOS_DELAY_WaituS(1000*MS_WAIT_WINDOW);
		ms -= MS_WAIT_WINDOW;
	}

	// wait() for the quotient
	PIOS_DELAY_WaituS(ms*1000);

	/* No error */
	return 0;
}
```

Not sure that that multiplication is improving things over the previous ```ms--```. We could try to get fancy with bitshifts, but I'm not sure it's worth it.

Thoughts?

kubark42, 2015-09-15:
Thinking of @tracernz's comment, we might also consider renaming this function to ```PIOS_DELAY_AtLeast_ms()``` and then this function is perfectly written as per its API promise. I'm leaning toward that as the best solution since looking at the code we don't have anything that requires a precise busywait. It's more like "ehh... 50ms ought to give the sensor time to warm up," or "yeah, that looks like the right LED flash rate".

mlyle, 2015-09-15:
This looks good-- it makes the problem more than 65x smaller.  :)

1000*MS_WAIT_WINDOW will evaluate to a constant and not be multiplied at runtime.  As to the ms*1000, that's just a single multiply instruction, which is single cycle on Cortex-M3 and later; plus perhaps a couple cycles to load the 1000 (perhaps not even that depending on how the compiler can schedule things)  Basically free.