TauLabs/TauLabs #1949: Higher level controllers should be implemented by callbacks, not tasks.

mlyle, 2015-09-21:
Right now there's a lot of resource usage for things like the path follower and altitude hold that happen even when the task isn't in use.  Some of this is probably unavoidable (keeping config around, etc), but...

Also we have lots of problems with sequencing of things like flight mode switches, waypoints, etc, which is preventing us from having a truly nested architecture of controllers.

Modules providing control functions should register function callbacks that then are used to call and explicitly engage the next step of control.  If absolutely necessary a future controller could use synchronization at this boundary and its own task... but the existing ones should be just fine being invoked this way.  Then we can do things like effectively share the stack between pathplanner, pathfollower, and altitude hold, too.

This also will contribute to/help us do things to allow picking different control strategies for different axes (e.g. unifying the altitude hold code or the yaw code in vtolpathfollower) in the future.

Trex4Git, 2015-09-22:
OP/LP already has callbacks functionality.
I had a look at it. In the developer part of the Wiki was also a description.

I think in principal for each task priority there is a "callback task".
And in each "callback task", different callbacks can by handled.

The Stabilization (outer Loop / inner loop also works with that at OP/LP)

mlyle, 2015-09-22:
Yah, I don't propose doing this globally, but just to control sequencing for the higher level flight modes (above attitude and rate loop).