TauLabs/TauLabs #1957: how to build a gcs release

wangzhenxing, 2015-09-23:
environment : windows7
I just run command "make gcs", copy "build/ground/gcs/" to new windows PC, run "build/ground/gcs/bin/taulabsgcs.exe". There is an error : lost Qt5Cored.dll.
I think it depends on qt environment.
@mlyle Is there any way to build a gcs release version ? or installable release version?
thanks

mlyle, 2015-09-23:
make standalone

or follow the directions to install nsis and then make package

closed 2015-09-23 by mlyle

wangzhenxing, 2015-09-23:
success, thanks
by the way, where can i get all the make params like : make all , make gcs, make standalone etc ?

tracernz, 2015-09-23:
Just run make with no target, it will give you descriptions of each target.

wangzhenxing, 2015-09-23:
run command "make"
it shows :
 mingw32-make: *** No targets specified and no makefile found. Stop.

mlyle, 2015-09-23:
In the taulabs source directory. :P

wangzhenxing, 2015-09-23:
it shows:

sh:C:\Program: No such file or directory
makefile:125; recipe for target 'help' failed
mingw32-make: *** [help] Error 127

but in makefile file,  i get all command

mlyle, 2015-09-23:
GNU Make does not support paths with spaces in them.  A development source tree probably doesn't belong in C:\Program Files, either.

http://savannah.gnu.org/bugs/?712

wangzhenxing, 2015-09-23:
resolved, thanks @mlyle 

move git from C:\Program Files(x86)\Git  to C:\Git

wangzhenxing, 2015-09-23:
and thanks @tracernz

wangzhenxing, 2015-09-23:
there is a new error by running command "make package"

!insertmacro: macro named "MUI_UNFUNCTION_DESCRIPTION_BEGIN" not found!
Error in script "f:/taulabs_branch_20150919_1/package/winx86/taulabs.nsi" on lin
e 364 -- aborting creation process
Makefile.winx86:17: recipe for target 'win_package' failed
mingw32-make[1]: *** [win_package] Error 1
makefile:1017: recipe for target 'package' failed
mingw32-make: *** [package] Error 2

is there any thing i must to config? @mlyle

guilhermito, 2015-09-23:
Please move this convo to the forum

wangzhenxing, 2015-09-23:
the forum is inactivity
no one reply

wangzhenxing, 2015-09-24:
I have got a plan, delete line 364 ~ 368 (drop uninstall function). And run command "make package", it works