TauLabs/TauLabs #1958: MPU6000 temperature calibration is incorrect

kubark42, 2015-09-23:
@AlessioMorale discovered this: https://bitbucket.org/librepilot/librepilot/pull-requests/45/lp-108-fix-temp-calculation-get-rid-of/diff

We suffer from it as well: 
* https://github.com/TauLabs/TauLabs/blob/next/flight/PiOS/Common/pios_mpu6000.c#L671
* https://github.com/TauLabs/TauLabs/blob/next/flight/PiOS/Common/pios_mpu6050.c#L704

From the MPU-6000 register map: 
> The temperature in degrees C for a given register value may be computed as:
```Temperature in degrees C = (TEMP_OUT Register Value as a signed quantity)/340 + 36.53```

mlyle, 2015-09-23:
To confirm, this isn't a correctness issue, but instead an ISR performance issue?  It looks like we set the registers to the right setting in flight/PiOS/STM32F4xx/startup.c:75 to preserve state.  (per http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0298a/index.html )

mlyle, 2015-09-23:
@Trex4Git this may relate to your observed CPU utilization differences between mpu6000 and mpu9250 at high sample rates.

mlyle, 2015-09-23:
Wait, in our code it doesn't even happen in the ISR.  I think this is a non-issue for us.

kubark42, 2015-09-24:
The constant is wrong. We use 35.0.

mlyle, 2015-09-24:
> Temperature in degrees C = (TEMP_OUT Register Value as a signed quantity)/340 + 36.53

vs

> 35 + (x+512)/340

are equivalent functions. 

Librepilot didn't change the function, they removed floating point.

AlessioMorale, 2015-09-24:
Sorry for the confusion, the main point of LP-108 is to remove the last, unneeded, floating point operation from the ISR. 
That part about the calculation was just a late night-induced brain fart :) then it was left like this for readability.

kubark42, 2015-09-24:
<facepalm>

I just noticed the constant in the math was different, I should have caught
that a component moved to inside the parentheses.

În data de joi, 24 septembrie 2015, AlessioMorale <notifications@github.com

mlyle, 2015-09-24:
:) It happens, assuming OK to close.

closed 2015-09-24 by mlyle