TauLabs/TauLabs #1959: Naze32 not getting past Board Identification in Setup Wizard

felixandersen, 2015-09-24:
Using RC-2 provided by @mlyle on the forum here: http://forum.taulabs.org/viewtopic.php?f=6&t=900

On the Board Identification page the "Detected board type" is **naze** but the "Next"-button stays gray making it impossible to continue. @fujin confirm the same behaviour and provided this screenshot
[image]

I am happy to build and test any pull requests on my board. Not using it for anything else at the moment.

fujin, 2015-09-24:
Confirmed verified. Screenshot is from my workstation. Thanks for logging. :+1:

mlyle, 2015-09-24:
Yah, the setup wizard is inop, as is firmware upload.

felixandersen, 2015-09-24:
@mlyle Is this something I could work on, with some pointers on where to start, or do you need to have intimate knowledge of the codebase?  Is the cause known?

mlyle, 2015-09-24:
@felixandersen We'd love a PR to fix this!  Absolutely. 

There are board-specific plugins in ground/gcs/src/plugins/boards_*  .

The setup wizard is in: taulabs/ground/gcs/src/plugins/setupwizard

I think the first problem is:

```
bool ControllerPage::isComplete() const
{
    return (getControllerType() != NULL) &&
           m_connectionManager->getCurrentDevice().getConName().startsWith("USB:", Qt::CaseInsensitive);
}
```

That is, we refuse to run the setup wizard for boards we're not speaking to with USB HID; Naze only does serial or USB-serial..  We should probably skip that second test for boards that don't have real USB.  

There are likely to be subsequent problems in the wizard.  We may not have appropriate defaults in the board plugin for different radio types, etc.  The automatic firmware update that the wizard tries to do won't work.  And I'm not sure about the reset midway through the process.

tracernz, 2015-11-26:
Fixed in #2036.

tracernz, 2015-11-26:
Can't close this since I removed my TL access.

closed 2015-11-26 by peabody124