TauLabs/TauLabs #1966: other languages ​​, text for translators missing

LinuxJohannes, 2015-09-25:
Hi.
working on the translation files, in my case, to translate it to german.

Some Text-strings are hard coded in the code. In this case, this text isn't included in the translation files.

Example 1 : configvehicletypewidget.cpp
```
    // Set up vehicle type combobox, don't change the order
    m_aircraft->aircraftType->addItem(tr("Fixed Wing"), AIRFRAME_FIXED_WING); // Index 0
    m_aircraft->aircraftType->addItem(tr("Multirotor"), AIRFRAME_MULTIROTOR);
    m_aircraft->aircraftType->addItem(tr("Helicopter"), AIRFRAME_HELICOPTER);
    m_aircraft->aircraftType->addItem(tr("Ground"), AIRFRAME_GROUND);
    m_aircraft->aircraftType->addItem(tr("Custom"), AIRFRAME_CUSTOM);
```
added tr(), text now in in the translation files. But in
``` void ConfigVehicleTypeWidget::doSetupAirframeUI(int comboboxIndex) ```
is a text verify which no longer works. Changes the code here, but, if somone will change the enum it wont work in future.

```
     // Check which tab page is currently selected, and get the item data from the appropriate combobox
-    if (m_aircraft->aircraftType->currentText() == "Multirotor"){
+    if (m_aircraft->aircraftType->currentIndex() == AIRFRAME_MULTIROTOR){
+//    if (m_aircraft->aircraftType->currentText() == "Multirotor"){
         frameType = (SystemSettings::AirframeTypeOptions) m_aircraft->multirotorFrameType->itemData(comboboxIndex).toUInt();
     }
```

Thats only one example.  Whats the best way to fix it?

kubark42, 2015-09-25:
I'd say your proposal is already a code improvement, even before considering translations. When enums are changed, we will get a compiler-time failure if any enums are left unchanged. Very easy to detect and fix. With a string compare-- such as is currently being done--, we might not be aware of breakage for a long, long time. So switching from string compares to enums is already an ideal fix.

In other words, @LinuxJohannes, you're on the perfect track. Thanks for looking into this!

LinuxJohannes, 2015-09-25:
Ok, thats easy to fix and will do it.

But there are another example in configfixedwidget.cpp
```
void ConfigFixedWingWidget::setupUI(SystemSettings::AirframeTypeOptions frameType)
{
    Q_ASSERT(m_aircraft);
	
    if (frameType == SystemSettings::AIRFRAMETYPE_FIXEDWING) {
        // Setup the UI
        setComboCurrentIndex(m_aircraft->aircraftType, m_aircraft->aircraftType->findText("Fixed Wing"));
        setComboCurrentIndex(m_aircraft->fixedWingType, m_aircraft->fixedWingType->findText("Elevator aileron rudder"));
```
and later in the file
```
    // Setup the appropriate aircraft type
	if (m_aircraft->fixedWingType->currentText() == "Elevator aileron rudder" ) {
        airframeType = SystemSettings::AIRFRAMETYPE_FIXEDWING;
        setupFrameFixedWing( airframeType );
	} else if (m_aircraft->fixedWingType->currentText() == "Elevon") {
```
Think, we need to change the kind of coding

kubark42, 2015-09-25:
Ick. I wonder if I'm not to blame for a lot of that back in 2011, and I've been slowly fixing it ever since. I think your proposed pattern will be a good upgrade for all this brittle code.

LinuxJohannes, 2015-09-25:
No problem. I'm not so good in c++ and QT. You can help to upgrade the code?

kubark42, 2015-09-25:
@LinuxJohannes We love helping people! You obviously know more than I do when I started out, since you're able to pick out my old mistakes. ;)

If you'd like to put together a PR I'll be happy to advise and test as much as you need. Getting translations going is exciting.

LinuxJohannes, 2015-10-23:
Hi.
Around 75 % of german translation is done. The left 25% are deeper in the GCS. Like to start the first PR to add the current work.

LinuxJohannes, 2015-10-28:
OK. Here my changes. Upps, cant attach files ?? "Attaching documents requires write permission to this repository. Try again with a PNG, GIF, or JPG. "

Want to attach the .diff files

kubark42, 2015-10-28:
@LinuxJohannes, if it's not too big you can go to gist.github.com and paste in the diff. 

But if it's really big (>>100 lines), maybe the best way to get us the changes is to make a git commit and then link to it?

LinuxJohannes, 2015-10-29:
OK, done. Here my changes. Discussion can be started.
https://gist.github.com/26d97bd82a751608d163.git

kubark42, 2015-10-29:
Thanks!

Looking at the diff, I'm going to recommend that this be split into two PR's, one to provide the translation table, and one to update the Qt code to use the ```tr()``` tag.

A note to other devs. When working on new PRs, think about fixing the ```tr()``` tags when you see them. Similarly, I'll try to be better about catching this in review so that we don't accidentally undo @LinuxJohannes's hard work.