TauLabs/TauLabs #1969: Discussion around Chebyshev type-II filtering for both D-term and gyro cutoff vs. Butterworth, current implementation

fujin, 2015-09-25:
http://eelinux.ee.usm.maine.edu/courses/ele486/docs/Chebyshev%20II.pdf
https://github.com/hrrr/ChibiFlight/blob/master/source/filter.c

from @Trex4Git on IRC:
> 22:43 <trex_> From the naming he uses a second order Butterworth lpf. If we want also go in that direction I think a second order chebyshev type 2 might be better
> 22:45 <trex_> Disadvantage of both tuning /setting the filter in the flight controller would not be so easy, so external calculating the filter coefficients and maybe having some selectable presets might work
> 22:46 <trex_> For the gyro filtering a higher order chebyshev filter could also reduce the delay

mlyle, 2015-09-26:
So we have no problems that we can attribute to poor d-term rolloff behavior.  So this probably doesn't deserve too much effort.

fujin, 2015-09-26:
@mlyle what about using either one for gyro cutoff to reduce delay per @Trex4Git ?

mlyle, 2015-09-26:
It's not clear to me that reduces the real delay.  And you've disabled that filter anyways in your flight.

tracernz, 2015-09-26:
It would help to define the problem(s) with the existing filter. Is it the delay you're concerned about, is the roll-off not steep enough? 

Re: the derivative filter: @borisbstyle has just done a lot of work to add the same derivative filter to CF and showed it made a large difference.

mlyle, 2015-09-26:
@tracernz Do you mean the same filter as us (first order IIR?)

tracernz, 2015-09-26:
@mlyle Yes: https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/pid.c#L214, https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/filter.c

borisbstyle, 2015-09-26:
It id IIR, but with nice human readable and flexible coefficient
adjustments.

mlyle, 2015-09-26:
@borisbstyle  Cool.  Our autotune procedure calculates the derivative cutoff coeff.

tracernz, 2015-09-26:
@borisbstyle TL filter cutoff settings are also in human readable format (Hz) :)
https://github.com/TauLabs/TauLabs/blob/next/shared/uavobjectdefinition/stabilizationsettings.xml#L35

borisbstyle, 2015-09-26:
Didnt realise that. Well guees both do the same than :)

How does the autotune deal with finding  right value? There is some kind of
noise threshold set?

mlyle, 2015-09-26:
Yes, effectively:

```
        double tau_d_roll = (2*damp*tau*wn - 1)/(4*tau*damp*damp*wn*wn - 2*damp*wn - tau*wn*wn + exp(beta_roll)*ghf);
        double tau_d_pitch = (2*damp*tau*wn - 1)/(4*tau*damp*damp*wn*wn - 2*damp*wn - tau*wn*wn + exp(beta_pitch)*ghf);

        // Select the slowest filter property
        tau_d = (tau_d_roll > tau_d_pitch) ? tau_d_roll : tau_d_pitch;
```

peabody124, 2015-09-26:
I should write up the derivation of the autotune calculation,  but yeah it calculations the cutoff for the derivative component based on the high frequency noise properties and desired response profile.

A paper on the system identification estimation should be coming soon :)

fujin, 2015-09-26:
@mlyle with regard to having disabled that filter in my flight, sure, presently, but under the consideration of the yet-to-be-named broadband preset for gyro downsampling, we might re-introduce some gyrocutoff, so was thinking about it in the context of that.

Trex4Git, 2015-09-26:
Just some examples for the Chebyshev Type II LPF. I did already earlier some investigations on it, but had no time for developing/testing yet. 
But after I have seen the discussion on IRC I think there might be some interest in it:

- F_stop = 200Hz, Dampening at F_stop: 20dB

Chebyshev Type II LPF:
- F_sampling=1.6kHz, FilterOrder=2 
 - delay@0Hz: 2.26ms / delay@50Hz: 2.87ms 

- F_sampling=1.6kHz, FilterOrder=4 
 - delay@0Hz: 1.62ms / delay@50Hz: 1.77ms 

- F_sampling=8kHz, FilterOrder=2 
 - delay@0Hz: 2.38ms / delay@50Hz: 3.02ms 

- F_sampling=8kHz, FilterOrder=4 
 - delay@0Hz: 1.70ms / delay@50Hz: 1.86ms 

The current 1tab IIR LPF:
- F_stop = 200Hz, Dampening at F_stop: 20dB (F_cutoff needed for this:19,5Hz)
- F_sampling=1.6kHz
 - delay@0Hz: 7.84ms / delay@50Hz: 0.74ms 

You can see that our current LPF has for similar filtering behavior significant higher delay and also the frequency dependency is not so nice (highest delay at 0Hz, and for higher higher frequencies smaller delay).
But it can be easy tuned in the flight controller.

Tuning the Chebyshev Type II LPF on flight control might be a lot of work to do, but with external tools (like Matlab , or Python) the coefficient calculation is also relative easy.
So when we have some fixed setpoints (sampling frequency, filtering behavior,..) we could store some presets on FC.

This filter also works nicly direct at 8kHz.
It has only ~0.1ms more delay than working on 1.6kHz, and no more CIC filtering would be needed (1. order CIC from 8kHz to 1.6kHz has 0.32ms delay, 2nd order 0.5ms). 
So ~0.2/0.4ms advantage, but I would expect higher CPU load, when a higher order Chebyshev is run at 8kHz. 

So maybe after testing the down-sampling, we can find  a control frequency at which there is no more advantage and only more CPU load (e.g. 1.6kHz vs 4kHz). We could than define some some cutoffs (like the LPF in the MPU) and store the coefficients for the filter in the µC.

mlyle, 2015-09-26:
Yah, but those are highly exaggerated cases @trex4git.  No one is running a 19.5 Hz cutoff.   Right now even for very large craft (where reducing delay is not such an advantage) 30Hz is fine.

> You can see that our current LPF has for similar filtering behavior significant higher delay and also the frequency dependency is not so nice (highest delay at 0Hz, and for higher higher frequencies smaller delay).

That's bad?  Frequency dependent delay is not so bad.  And low delays at higher frequencies means more phase margin.

We're talking an awful lot about something that there's very, very little reason to believe will improve anything.  Twitchy little craft do fine without the filter now.

fujin, 2015-09-29:
Super cool :) closing this out for now, can be re-opened post gyro-downsampling (if we continue going in that direction)

closed 2015-09-29 by fujin

Trex4Git, 2015-09-29:
My data was just an example,  I started with a setup for the chebyshev to do the comparision.
After that, I tunded the current LPF to have a similar behavior.

So the 19.5HZ are maybe not realisatic for a gyro LPF, but in principal for higher cutoffs the chebyshev  keeps its relative advantage in delay I think.
But absolutly the difference might be to small to really notice it for a real and working setup.

But I agree that closing that for know is fine. We maybe could optimize the gyro filtering if we have have knowledge what control frequency gives an advantage, and what gyro filtering is needed for that.

I just wanted to share that info, if someone is interested.