TauLabs/TauLabs #1970: Thoughts on mixer infrastructure

mlyle, 2015-09-26:
So one thing we could do is change the mixer to a small RPN program.  The idea is we'd use 8 bit opcodes, and could have encodings like:

```
0000xxxx push variable onto stack
      x=0 to 6 =  throttle, roll, pitch, yaw, accessorydesired0, ...
      x=16 to 25 = an existing, already completed mix for output channel x-16
0001xxxx store output channel x's value, scaled -1 to 1.  
001xxxxx evaluate function/operator.
      x=0 add; pop 2 elements from stack, add them, and push result
      x=1 subtract
      x=2 multiply
      x=3 divide
      x=8 powf
      x=9 sinf
      x=10 cosf
      x=11 atanf
      x=16 interpolate using curve1
      x=17 interpolate using curve2
      x=18 pop all values in stack, add, and push (hack)
010xxxxx load low 5 bits of constant, store completed constant onto stack
011xxxxx load low 5 bits of constant, pop stack, multiply, and store (hack)
1xxxxxxx load high 7 bits of constant (fixed point format TBD)
```

So an existing quadcopter mix for the first channel would look like:

```
Load pitch
LoadHiConst
LoadLowConst-Multiply
Load roll
LoadHiConst
LoadLowConst-Mutiply
Load yaw
LoadHiConst
LoadLowConst-Mutiply
Load throttle
Apply throttle curve
AddAll
Set output 0
Load pitch (for next channel)
...
```

So this results in 13 bytes per channel, versus 5 currently or 10 if we double precision.  Unfortunately it only increases our precision to 12 bits, and presumably we'd want to put one or two of them left of a decimal point.  Probably not quite acceptable.  Then again outputs right now have 10 bits precision.

kubark42, 2015-09-26:
Neat idea. How do we always verify that the stack is sufficiently computationally efficient, no matter what the user asks of it?

Whatever we do, we should make it easy to add additional mixers. Fixed wing flight requires a bunch of other state inputs to the mixer, such as flaps, spoilers, landing gear, etc...

mlyle, 2015-09-26:
So there's no looping or conditional directives.  So if you can bound the time per "instruction" and the max number of instructions you can bound instruction time.

kubark42, 2015-09-26:
> So there's no looping or conditional directives. So if you can bound the time per "instruction" and the max number of instructions you can bound instruction time.

Won't referring to the curve interpolation somewhat violate this?

> x=18 pop all values in stack, add, and push (hack)

Should this be 31, so that we reserve the rest of the space for other operators? I think that a ```log`` operator might come in handy.

mlyle, 2015-09-26:
So, curve interpolation, atan, etc, take awhile.  But even if your program is all curve interpolation, all atan, etc.. you still know the maximum runtime.