TauLabs/TauLabs #1971: i2c/usart openlog-compatible/blackbox logger

fujin, 2015-09-30:
I would like to have high rate (115200bps+) usart logging of some values to sdcard over i2c if possible.

tracernz, 2015-09-30:
I2C could be quite a good idea on where there is a dedicated I2C bus to the flexi port (e.g. Sparky2). You wouldn't want to saturate the I2C bus with data where it is shared with sensors though. Interestingly the Sparky and Sparky2 both have 10 k pullups on the flexi-port which would typically limit the speed to less than 400 kHz unless you add some smaller external pullups. You'd also have to modify the OpenLog firmware to support I2C instead of UART.

jihlein, 2015-09-30:
I'm going to try modifying the existing logging module to log to an openLog at 115200 baud.  Going to try it on the AQ32 (F4 based) hardware first, as it has no spi flash.  If it works, I'll see if it will work on the F3 based boards.  I'd rather not use I2C, as there is typically only 1 external I2C bus and I'm using that for external mag.

tracernz, 2015-09-30:
Even with their data compression Cleanflight are unable to log at full rate over 115200 baud UART, UAVTalk is MUCH larger.

kubark42, 2015-09-30:
@fujin, is 115,200bps real-world throughput? Or the bus speed itself?

The logging rate we get with #1928 is on the order of 1-5MB/m. So with all respect to @tracernz, we log a sizable amount of data and find that 115,200bps is enough for significant insight into the sensors (every single measurement) and vehicle state (~50Hz for most relevant UAVOs).

I like the idea and would welcome PRs extending logging functionality. This is an area where we can grow significantly with very little downside.

tracernz, 2015-09-30:
Ah, yes. I was thinking of a blackbox which records absolutely _everything_, like Cleanflight. If you only log selected fields or UAVOs the requirements can be much lower. :) UART is certainly simpler so if it works for what we want to do then it's a pretty clear winner.

mlyle, 2015-09-30:
We should consider doing it at the uavtalk layer, though.  If we add incremental encoding, etc, it could benefit over-the-radio interfaces too.

mluessi, 2015-09-30:
@tracernz regarding logging in cleanflight, the difference is that "everything" in cleanflight is far fewer variables than the entire set of UAVOs in TL. 

Using [openlog](https://www.sparkfun.com/products/9530) hardware on a serial port with 115200bps with the current or improved (#1928) logging module should be fairly trivial, as the interface to the internal logging file system acts like a serial port. One would just need to modify the way the files are opened / closed.

Logging to the SPI flash can be fairly fast (I have measured about 15kB/s = 120000bps). One problem is that erasing a page takes a while (about 50ms), so unless the file system is formated, it will be slower. If the external logger can sustain the 115kbps throughput, it may be faster than the current logging as we don't have to wait for pages to be erased.

mluessi, 2015-09-30:
One thing we could think about to increase the throughput is to reduce the size UAVTalk header used for logging (see [spec](https://github.com/TauLabs/TauLabs/wiki/Development-UAVTalk-Protocol)). Currently, we have either 11 or 13 bytes header overhead per object, which is about the same size as the payload for things like gyro data (12 byte payload). It should be enough to use a packet only consisting of (Object ID, timestamp, instance ID (optional), payload, CRC), so either 7 or 9 bytes overhead per packet. It will make log parsing a bit more difficult, but the space savings may be worth it.

jihlein, 2015-10-01:
I've got the AQ32 sending logs to the openLog.  Logs are being created, but the python scripts are not working.  Don't know too much about python, I'll try Matlab tomorrow.  If it works, I'll clean things up a bit and post a PR, not for merging, but for further dicussion.

mlyle, 2015-10-02:
@jihlein if you put the logfile somewhere I can look at it with the python scripts and see if I can tell what's wrong.

It may just be that you need to pass -t to the dumplog script.