TauLabs/TauLabs #1973: Use mass storage for retrieving logged data

kubark42, 2015-10-01:
It would be very nice to use a mass storage driver for accessing the logged files. This might dramatically speed up logfile downloads. I'm thinking several MB/second instead of several seconds/MB. 

Obviously, we'd have to FAT format the EEPROM like an SD card. Or we'd have to have a on-the-wire translator so that the system thought it was accessing a FAT filesystem. I've got experience doing this on Atmel with FAT 12/16, it worked quite well.

What are potential traps and pitfalls in doing this?

See #1952 and #1971, etc...

mlyle, 2015-10-02:
> Or we'd have to have a on-the-wire translator so that the system thought it was accessing a FAT filesystem.

I think I'd prefer this.. As little as our NOR flash is, it's nice for it just to be a nice flat buffer for logs that we can pre-erase, etc.

kubark42, 2015-10-02:
Short term, this makes more sense to me as well. It reduces flight-time overhead for using the EEPROM and allows for more easy hacking of sectors, like we do with waypoints.

Long term, I think a formatted filesystem makes more sense. If allows for choosing between SD card and NVRAM to be a hardware-only question, and it means that we don't have to worry about pre-allocation of space.

As an aside, I believe that FAT12/16 is pretty efficient. I don't recall there being any overhead past the first few sectors. We can format it and re-establish the first 512 bytes without too much difficulty. We could also reserve a section at the end of the EEPROM for the flat buffer.

michaelrempel, 2015-10-02:
When this is figured out I am very interested in obtaining log files to work on sensor improvements. I have a Brain which I understand limits my ability to jump on latest firmware. For logging would standard USB key work with FAT 16, or is that too much overhead code? I can see it would need an adapter.

mlyle, 2015-10-02:
@michaelrempel this is about the onboard NOR flash.  Right now you can enable logging to flash and use the GCS to download logs.  It's kinda slow to pull log files.

This issue is for the enhancement to support USB mass storage and "look like a disk drive" when plugged in, making it easier/faster to download logs.

mlyle, 2015-10-02:
@kubark42 One thing I wonder is if it would be possible to use the python API to make something that is quicker at downloading logs / what the bottleneck is.

kubark42, 2015-10-02:
@mlyle I believe it's the limitation of the HID interface, as per the discussion on ```bInterval``` in #1418. The best you can do is 64KBytes/second, which works out to ~15s/MB. Even more complicated, the logfile data comes in over the same bandwidth as the regular UAVOs so it competes with them. The upshot is lots of USB time spent for very little data, so I don't know how python could address the USB limitations.

Plus, pure python isn't user friendly. Don't mean to pick any battles here-- and I'm a big iPython notebook fan--, but most users will want to get this through a pretty UI. 

@michaelrempel, I'm happy there're more people on the sensor bandwagon! See #1928 for the key to unlocking recording of every single sensor packet. As an aside, @mluessi is pretty good about keeping BrainFPV up-to-date. Are you having issues where it's not as fast to update as you would like, or you're wanting to have the absolutely latest merges as soon as they land in ```next```?

mlyle, 2015-10-02:
> I believe it's the limitation of the HID interface, as per the discussion on bInterval in #1418.

Ahh.  That's probably true for HID, but a good CDC-ACM implementation shouldn't be so limited-- it's used for things like 3G modems that go several megabits per second. 

> Plus, pure python isn't user friendly. Don't mean to pick any battles here-- and I'm a big iPython notebook fan--, but most users will want to get this through a pretty UI.

Depends on the use case.  This is for developers mostly, right?  Log analysis in the GCS doesn't seem to work too well in practice, and so it seems like you end up dumping it into matlab or python anyways.

kubark42, 2015-10-02:
> a good CDC-ACM implementation shouldn't be so limited

So would you recommend using the ComBridge instead? It's supported on all platforms (noting that the Windows CDC driver is a PITA for users to install because it's not signed) and could function concurrently to the HID.

> This is for developers mostly, right?

I actually think of this as a user-facing feature. There are a lot of ArduPilot automated loggers which helped the community track down bugs and poor configurations. There's also a lot of users who want to see GPS tracks and other telemetry data. 

That's not to argue with your point that right now it is a developer feature and principally-- exclusively?-- used by people who then import it into python or matlab.

mlyle, 2015-10-02:
> So would you recommend using the ComBridge instead? It's supported on all platforms (noting that the Windows CDC driver is a PITA for users to install because it's not signed) and could function concurrently to the HID.

I think it's something we could do short term to alleviate pain for us, while we wait for those better user-facing tools to exist.

tracernz, 2015-10-02:
I have thought about this in past too but not had any time to look further into it. The idea is very good in principle. 

Re: Python vs. GCS: It would be great to eventually have a nice log parsing/display module within GCS, similar to the Cleanflight Blackbox log tools perhaps.

peabody124, 2015-10-03:
I'd have to review the FAT specifications but I think having that work in place would have a substantial memory requirement (at least be able to buffer a sector) as well as a performance hit for erasing sectors multiple times. If it were possible to do an one-the-fly translation that would probably have some efficiency benefits on the logging side of things.

Regarding download speeds I would ack what @mlyle said that VCP would be tons faster than HID (which specifically implements an input device and isn't a bandwidth optimized USB protocol). This is true even when using telemetry.

@tracernz regarding visualization there are already a number of python scripts I've uploaded that will create video overlays based on log files like what i've used in my own videos.