TauLabs/TauLabs #1988: Setup Wizard: Add selection of default gains by vehicle size

tracernz, 2015-10-12:
One thing that is very apparent from @borisbstyle's Cleanflight fork is that people very much like a set of default gains, filter cut-offs etc. which match their vehicle size/configuration (in that case the default settings are targeted at 250 racing quads). It would be nice to add selection of vehicle configuration to the setup wizard which would allow different initial settings for a small set of different vehicle sizes. This would then allow autotune to be more effective on the first pass.

fujin, 2015-10-12:
Definitely make sense to me, I go to like: 1khz mpu rate, 184hz mpu gyro dlpf, 0hz gyro cutoff then try to establish some baselines, cycle until I get ident where it looks good and flies nice. @mlyle had similar ideas longer term for a 'broadband mode' based on the outcomes of the gyro downsampling work @ https://github.com/TauLabs/TauLabs/pull/1781

tracernz, 2015-10-12:
It would also allow higher default gains for larger craft, whereas our current default gains are limited by small overpowered vehicles and are much too low for large machines with lower power-to-weight ratios and longer arms.

fujin, 2015-10-12:
Yup, that's definitely sensible too!

kubark42, 2015-10-13:
Seems sensible. It would be very easy to ask the user what the arm length is and then use that to set the default.

peabody124, 2015-10-13:
Also as a practical short term approximation, having people exchange .uav files containing a minimal set of values (e.g. stabilization, flight modes, actuatorsettings) gives an easy way to test things