TauLabs/TauLabs #1991: Uncrustify CI check

kubark42, 2015-10-15:
Could Jenkins run uncrustify and fail the PR if the style rules aren't met?

Maybe it could do it only on the lines being changed, or perhaps just the files that are touched? 

P.S. If we are going to uncrustify the whole project-- which we should do!--, it should come in as uncrustify only commits. Otherwise it makes tree merges harder than they need to be.

PTDreamer, 2015-10-15:
So you are also proposing moving from astyle to uncrustify?

tracernz, 2015-10-15:
@kubark42 I ran style over all of the flight code a little while back just playing around, you can see the diff here: https://github.com/tracernz/TauLabs/commit/19df167912ceb1a5fae720bb687a2a9b83c357ff

kubark42, 2015-10-15:
@PTDreamer, I thought ```astyle``` was the tool and "uncrustify" was the action.

PTDreamer, 2015-10-15:
No two different tools.
http://astyle.sourceforge.net/
http://uncrustify.sourceforge.net/

kubark42, 2015-10-15:
@PTDreamer Changed the OP to reflect this.

Now I'm annoyed. "Uncrustify" was so useful as a verb. "Clean up style differences according to guidelines" is a lot harder to say!

@tracernz Thanks!

kubark42, 2015-10-15:
@tracernz, changes like https://github.com/tracernz/TauLabs/commit/19df167912ceb1a5fae720bb687a2a9b83c357ff?w=1#diff-e5485a5f7fa253ced2c1845c27ebd166R453 look wrong. The brace shouldn't be on the next line when defining arrays, right?

PTDreamer, 2015-10-15:
> @PTDreamer Changed the OP to reflect this.
> 
> Now I'm annoyed. "Uncrustify" was so useful as a verb. "Clean up style differences according to guidelines" is a lot harder to say!

Nahh actually uncrustify looks better to me, we just need to mod the config file to get the style we want.
So we should start by deciding the tool we want to use.
About CI integration that is very straightforward both as a warning or as merge preventing error. For me the current issue is that with the current project changes taking place we will probably change the current CI infrastructure.
Making the chosen tool go trough the changed files only shouldn't be that hard either, it will probably take same git and grep foo only.

mlyle, 2015-10-15:
https://github.com/d-ronin/dronin/pull/22/files has the proper uncrustify config file

Basically, I chose something that was a compromise that matched most of the code / moved the fewest lines.  There's a bunch of areas we use inconsistent indentation (continuation lines, switch/case statements), so for the things that were in question I just picked what looked like it was most frequently used.

tracernz, 2015-10-15:
> @tracernz, changes like tracernz@19df167?w=1#diff-e5485a5f7fa253ced2c1845c27ebd166R453 look wrong. The brace shouldn't be on the next line when defining arrays, right?

@kubark42 There are a few things I'd consider less than optimal in my cursory glance but that particular one does conform to the kernel style. It states that the closing brace is always first on the line. Of course our code doesn't strictly follow the kernel conventions and especially in the case of the non-spacing/indentation type style rules, e.g. "Chapter 20: Conditional Compilation" which we violate badly right throughout.