TauLabs/TauLabs #1996: Determe where a queue is created

kubark42, 2015-10-21:
Queues don't have any knowledge of where they were created:

```
#if defined(PIOS_INCLUDE_FREERTOS)

struct pios_queue
{
	uintptr_t queue_handle;
};

#elif defined(PIOS_INCLUDE_CHIBIOS) 

#include "ch.h"

struct pios_queue
{
	Mailbox mb;
	MemoryPool mp;
	void *mpb;
};

#endif /* defined(PIOS_INCLUDE_FREERTOS) */
```

and this leads to problems when the event system detects that a queue isn't being completely emptied because it's hard to determine where the guilty queue is, and whether it's always the same queue.

I propose adding an identifier-- perhaps just four characters-- which can be used to figure out which queue is struggling. 

It doesn't seem like this would impact anything beyond static memory load. We could do it for as cheaply as 1 byte, since it's unlikely we'll have 256 threads in the near future, but then we would need a look-up table to relate the code to the thread. I would prefer to use 4 characters, which will give us enough textual information to figure out the problem's origin.

However, mucking around with the queue system is getting arm deep into the foundational code, so we should tread lightly.

Thoughts?

mluessi, 2015-10-21:
The UAVO manager already records `lastQueueErrorID`, maybe we could just expose that via system alarms, so one would know what type of UAVO caused the overflow. It wouldn't directly indicate the offending queue, but would make it easier to track down. Alternatively, put a breakpoint in `sendEvent` in the UAVO manager and inspect the value there.

kubark42, 2015-10-21:
I believe that this information is already in SystemStats.ObjectManagerCallbackID and SystemStats.ObjectManagerQueueID.

mluessi, 2015-10-21:
I see.. and this isn't enough for tracking down the offending queue?

mlyle, 2015-10-21:
This looks like it needs some help in general.  On freertos the double indirection shouldn't be necessary and memory can be saved / performance can be gained.  

On chibios we could use __builtin_return_address to get the address of the code that called it to avoid having to fix up all the existing code.  Then in turn that address could be compared to listings or symbol table.  It's a little less convenient than a perfect 4 letter id but it's a lot less work.

mlyle, 2015-10-21:
P.S. We need to kill the use of uintptr_t everywhere as it's not type-safe.  If you conflate types you get no compile-time warning.

A pattern I like using is... in public headers:

```
/* anonymous struct declaration.  this is opaque */
struct something;
/* a something_t is a handle to a device/api/etc that is described by a struct something
typedef struct something *something_t ;

int my_call_to_do_stuff(something_t handle);
```

And then in the private header you actually describe what's in the struct something:

```
/* de-opaqueify this for modules that can see the private header */
struct something {
    Mailbox mb;
};
```

Further, whenever possible, you move the private headers to be next to the code so they're not in other modules' include paths-- so someone can't even be naughty and include the private header.

kubark42, 2015-10-21:
> I see.. and this isn't enough for tracking down the offending queue?

Sometimes, but frequently it's not. The problem I'm currently running into is where I have multiple modules which are all subscribed to the same UAVO. For instance, we can't disambiguate anymore between logging and anything else.