TauLabs/TauLabs #2001: Modify PIOS_HAL_ConfigurePort to use struct as argument

mluessi, 2015-10-22:
As discussed in #2000, the function `PIOS_HAL_ConfigurePort` currently uses a large number of arguments, which has some downsides. First, all callers of the function (=all targets that use HAL) need to be modified whenever something is added, even if the new functionality is not used by a specific target. Second, the huge number of arguments is becoming difficult to use (it is easy to put NULL at a wrong location etc.). Instead of using a large number of arguments, I propose something like

    void PIOS_HAL_ConfigurePort(HwSharedPortTypesOptions port_type,
    const struct HalPortOptions * port_options);

where `struct HalPortOptions` contains all the parameters needed. A caller would then use something like

    struct HalPortOptions opts;
    memset(&opts, 0, sizeof(opts));
    opts.pios_usart_cfg = my_cfg
    PIOS_HAL_ConfigurePort(my_type, &opts);

in case only `pios_usart_cfg` is needed for the port type exposed by a particular target.

mlyle, 2015-10-22:
Repasting my comment from #2000:

Really PIOS_HAL should not be changing so much. The most recent was my bad for not finding during review-- the logging settings should be accessed by the actual logging task and the way that works doesn't seem to make sense for the different targets (AQ32 looks at configuration for where log should go for flash vs. openlog, when AQ32 doesn't have flash... but the other targets don't? The HW target looks at a non-HW specific variable?) The main thing I want to do is remove some of the sbus, DSM, etc specific configuration that can be calculated at runtime so it's not such an argument-list-of-death.

The previous one was adding a fairly fundamental type (PWM). It highlights a vulnerability with how we do CI/PR building (something that has been open a week should be re-tested before merge... people who are doing test on PRs should be doing it merged or rebased to next).

mluessi, 2015-10-22:
But obviously it does change and I don't see why it wouldn't change in the future when more things are added. The huge parameter list makes it a real pain to maintain a target IMO.. (the error messages are quite cryptic).

Any reason why you would be against using a struct? IMO it's a much cleaner solution that makes maintaining targets easier.

mlyle, 2015-10-22:
I'm not absolutely against it.  But:

1. I am really hoping to cut down the huge parameter list so it's not so bad.  I don't understand why we have multiple versions of the serial configuration often-- I think it's mostly for F3 inversion but has become yuckiness that's infected all the targets.  These things can instead be calculated at runtime in PIOS_HAL.
2. I'm hoping that it won't change much in the future.
3. Compile-time checking that all things are provided / having targets break if we add something new so that we can decide whether it's optional or not for that target is nice, rather than them just being omitted.

Admittedly, it's a bit worse when we add a new, optional thing to PIOS_HAL... but that was more of an interim hack that is spreading than something that should happen frequently, I think.

jihlein, 2015-11-02:
Another point with PIOS_HAL:

Frsky Hub and Sport telem is broken on F3 targets because the Frsky specific serial port config is not passed to PIOS_HAL, the standard serial port config is.  While a long term solution for PIOS_HAL is discussed, I've verified adding another argument to PIOS_HAL with the Frsky Hub/Sport serial port config fixes the issue.  I've only made the change on my private branch, but can add to next if desired.