TauLabs/TauLabs #2002: #1759 uninitialized pointers

PTDreamer, 2015-10-22:
#1759 added two pointers to the FieldInfo_s structure (part of the uavo generator) which are used uninitialized causing segfaults.

mlyle, 2015-10-22:
I don't understand.  FieldInfo is allocated with the new operator with the default constructor from what I can tell:
```
    FieldInfo* field = new FieldInfo();
```

A struct default constructor sets the memory to zeroes in C++, doesn't it?

mlyle, 2015-10-22:
Do you have a backtrace?  If it's failing here:

```
    if (!field->parentName.isEmpty()) {
```

the issue is a missing null check there.

PTDreamer, 2015-10-23:
> A struct default constructor sets the memory to zeroes in C++, doesn't it?

My understanding (reading point 8.5 dcl.init of the standards) is that it doesn't.
Attached is the VS debugger screenshot and the dump file.

[image]

mlyle, 2015-10-23:
dcl.init 8.5.8.2, 11.. it's supposed to happen.  That being said, there's something obviously wrong.  You're building this with MSVC?  I don't have a MSVC environment unfortunately..