TauLabs/TauLabs #2003: OpenOCD configs are stale

kubark42, 2015-10-23:
OpenOCD 0.9.0 no longer works with the in-project config files. This is true for both the bootloader and the debug file. Fortunately, everything we need is already in latest openocd. ```stm32f4discovery.cfg``` for debugging and ```stm32f4x_stlink.cfg``` for the flashing tool. I'm happy to say that in both cases I am able to flash and debug with 0.9.0.

Apparently our current approach is Not Good(TM). From one of the OpenOCD maintainers:

> PaulFertser: I hope the OpenOCD project manages to balance between backwards compatibility and going forward.
> [4:25pm] PaulFertser: Some things just need to be broken. And popular adapters and targets just need to be upstreamed, then it would be maintained automatically.
> [4:25pm] PaulFertser: So using out-of-tree configs is sort of asking for trouble.

His point is that properly installed openocd already knows where to find its config files and we shouldn't try to set up the flashing operations ourselves. For the flashing tool, we need to "source [find target/stm32f4x_stlink.cfg]". For the debugging tool, I'm less sure how this should happen.

In both cases, I'm not familiar enough with OpenOCD to guess at what level these changes should occur. In my experimentations, I can replace all of stm32f4xx.stlink.cfg with 
```
#
# STM32f4x stlink pseudo target
#

source [find target/stm32f4x_stlink.cfg]
```

and flashing works. I don't know how this change should happen for the debugger side, though.

@solidgoldbomb, @lilvinz this is something you know more about than I think anyone else here. Got a second to comment?

tracernz, 2015-10-23:
Personally I just run `opened -f interface/stlink-v2.cfg -f target/stm32f4x.cfg`, no in-tree config files required, and you can just change f4x for f3x or f1x as required. `target/stm32f4x_stlink.cfg` is deprecated so probably best we don't use that one.

paulfertser, 2015-10-24:
There's absolutely no difference config-wise between flashing and debugging. _stlink configs are deprecated but if you really need something that would work with 0.7.0 as well as with 0.9.0, that's probably a reasonable option.

Another option is to direct the users towards OpenOCD README and let them run it however they want, e.g. some people will use other JTAG/SWD adapters etc.

If your firmware might go to sleep mode or remap SWD pins, you can physically connect the SRST line and use "reset_config srst_only connect_assert_srst" to be able to attach to the target no matter what.

PS: I'm not an OpenOCD maintainer.