TauLabs/TauLabs #2004: ConfigAutotuneWidget::recomputeStabilization has hard-coded convergence limit

dustin, 2015-10-23:
There's suspicious that 30 minute not be the right value and can lead to bad results.

https://github.com/TauLabs/TauLabs/blob/next/ground/gcs/src/plugins/config/configautotunewidget.cpp#L264-L271

mlyle, 2015-10-23:
Autotune definitely does things that are obviously unconverged and jumps between random values at either low taus or low damping factors.  Also it has no problem returning solutions with negative natural frequencies and negative outer loop Kp.

peabody124, 2015-10-24:
If you could post a screenshot of an AT where it produces bad natural frequency that would help. I'd especially like to distinguish if it was problematic inputs (e.g. extremely low taus) versus a convergence problem. I would have to go back to the derivation but there were a few heuristics that were required to come up with a closed form solution (e.g. the relative placement of the two poles for the rate response) that are probably only appropriate within a certain domain of parameters.

mlyle, 2015-10-24:
> like to distinguish if it was problematic inputs

Problematic inputs definitely do it-- e.g. valid tau but moving the slider left:

[image]

I've also had tunes where the tau and values seem reasonable, but moving the slider around causes natural frequency and output values to jump seemingly randomly.  I am guessing this is a convergence problem.